motioneye-project/motioneye #2103: Wrythe Green Surgery

ronsteenbergen, 2021-04-20:
Hi there,

Is there a way to set the time in MotionEye when the Pi is not connected to the internet?

starbasessd, 2021-04-20:
MotionEye, yes in crontab in your base os, you could do ifup/ifdown type commands.
MotionEyeOS, yes, sort of, but involves disabling network watchdog, also, as losing network connectivity causes panic reboots by design.
You can also use Working Schedule to prevent motion detection and notifications, as long as it doesn't cross Midnight.
Some routers/WiFi routers can disable specific ports (ethernet) or device traffic (kid controls) also.

starbasessd, 2021-04-21:
Yes. You need to console or ssh into the Pi running motionEyeOS and run the command:
date -s "Thu Aug  9 21:31:26 UTC 2012"  << DayofWeek Mon Day 24HourTime TimeZone Year

ronsteenbergen, 2021-04-22:
thanks, it works

closed 2021-04-22 by starbasessd