motioneye-project/motioneye #2104: raspberry pi with system default python3

questrov, 2021-04-21:
Installed circuit python [https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi](url) which changes system default to python3, which breaks motioneye.  

Bummer since servokit [https://learn.adafruit.com/adafruit-16-channel-servo-driver-with-raspberry-pi/using-the-adafruit-library](url) drives a pwm board nicely with incremental servo control similar to servoblaster which does not seem to work on a Pi4 (works great on zero) along with all the other Adafruit gizmo's.

Does the motioneye python3 fork work on Pi4 yet or is there another workaround?  If so are there any details how to get there from here?  

Tried pip2 on a fresh OS install with circuit python but didn't do something right as the motion service would not start, gave same same error as when I first switched to system default python3...

starbasessd, 2021-04-21:
Please install motion/motioneye from the github motioneye wiki for either Ubuntu 20.04 or RaspberryPiOS. That will install the correct pip2. You will have to do python2 and pip2  commands for motion/motioneye or set default to python2 and call circuit python with python3 commands.

questrov, 2021-04-27:
Sorted thank you very much!

closed 2021-04-27 by questrov