motioneye-project/motioneye #2107: Frame changes doesn't appears.

ALF-2021, 2021-04-22:
Good morning i have 2 TAPO C310 install on motioned in Home assistant but when i set "Show frame changes" on motioneye on the right corner of cameras appeares only - and don't work the motion trigger.

Can you help me?

Thanks.

starbasessd, 2021-04-22:
It should show a red line box around where it sees motion.
Try lowering Settings, Motion Detection, Frame Change Threshold to 0.1%, then slowly move it up if it is too sensitive.
I do not recommend changing Maximum Change Threshold, or enabling Auto THreshold Tuning.
I have Auto Noise Detection ON, Light Switch Detection set to 0% and Despeckle Filter set to On
For most of my cameras I have 
Motion Gap 2 seconds
Captured Before from 2-10 frames
Captured After set the same
Minimum Motion Frames set to 2 frames

ALF-2021, 2021-04-22:
I put your setting in Motioneye but the camera don't reach the motion.

[image]

[image]

starbasessd, 2021-04-22:
If you install RaspberryPiOS, then follow the instructions for installing
motion/motionEye here:
Install On Raspbian · ccrisan/motioneye Wiki
<https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian>
Do Not install Home Assistant.
Test for motion detection working.
Let me know.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ALF-2021, 2021-04-22:
There isn't another way to check it motion detection works?

starbasessd, 2021-04-22:
It is known to work very well.
Your Pi is working
Your Camera is working fine, you are getting a good picture.
Most of motion/motionEye is working. THere are a lot of known issues with
the motioneye plugin modified bt FRENCK. The quickest way to check is to
take HomeAssistant out of the picture. You don't have to use your current
SDCard, you can test with another one.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ALF-2021, 2021-04-22:
Ok. I will do and I will send you the feedback. 

Thanks. Best regards.

ALF-2021, 2021-05-18:
Hello, I updated HA and reinstalled one cam. 
I noticed the number of pixel on high right corner in on, but when I turn on "working schedule" it disappears. Check the images. [image]

[image]

Can you help me. 

Best regards.

starbasessd, 2021-05-18:
Is the time and date set correctly for the OS?
If you want motion to be detected, using motionEye's working schedule, you need to set the start & end times for either detect or outside working schedule. 
I really wish you would take HA out of the picture, to see if it's a motionEye issue or HA issue.

ALF-2021, 2021-05-18:
I'm sorry but I don't understand which you mean with:

"I really wish you would take HA out of the picture, to see if it's a motionEye issue or HA issue."

Can you explain me better please?

Thanks.

starbasessd, 2021-05-18:
https://github.com/ccrisan/motioneye/issues/2107#issuecomment-825167117

ALF-2021, 2021-05-18:
I'm sorry you were joking. 

Do you need the side with HA settings?

starbasessd, 2021-05-18:
No. There are many cases here where running with HA has problems that go
away when not using HA.
Not joking.
Maybe someone else will be able to help with HA, or even the HA support at

Home Assistant Community Add-on: motionEye - Home Assistant OS - Home
Assistant Community
<https://community.home-assistant.io/t/home-assistant-community-add-on-motioneye/71826>

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ALF-2021, 2021-05-18:
Ok I will try with Motioneye OS. 

Thanks.

JamBalaya56562, 2026-11-08:
Closing, as this was with the Home Assistant add-on of 2021 and the reporter moved on to try motionEyeOS. The changes shown by Show Frame Changes are counted by Motion only while motion detection runs, and Working Schedule pauses detection outside its hours, so the number going away then is expected. If a camera with detection running still shows no changes on a current release, please open a new issue with `motion.log`.

closed 2026-11-08 by JamBalaya56562