motioneye-project/motioneye #2108: webcam disconnected - deleted files

beppe84, 2021-04-23:
the webcam was disconnected by mistake, the photos and recordings are no longer present. is there any way to get them back?

starbasessd, 2021-04-23:
You can look for the files from the OS side.
Are you using motion/motionEye on an OS like Raspbian, or motionEyeOS?

beppe84, 2021-04-23:
i'm using motionEyeOS v. 20200606 running on a raspberry pi3

starbasessd, 2021-04-23:
If you have a monitor and keyboard attached, sign in as admin, and type the following commands:
cd /home/ftp/find ./ *.jpg     # and look for the missing snapshots
find ./ *.mp4   # or whatever format you are using for video
you can also run the commands in /data 
If you have ssh access, then do the same thing there.If you need more complete instructions, let me know
ps. This is the motioneye github, oriented to motion/motioneye on another OS, there is a motionEyeOS specific github too.

JamBalaya56562, 2026-11-08:
Closing, as this was answered: a disconnected camera does not delete its files, and on motionEyeOS they can be searched for under `/data` and `/home/ftp` as starbasessd showed. If they are not there, they were removed by the OS or the card, not by motionEye; motionEyeOS itself is maintained in its own repository (motioneye-project/motioneyeos).

closed 2026-11-08 by JamBalaya56562