motioneye-project/motioneye #2109: Motion eye no longer runs after 1st reboot

zgadgeter, 2021-04-24:
Hi, i am running motioneye in a container in proxmox. First time after install it boots up nicely and runs fine. But after a reboot it no longer seems to run. I can ssh into root, but the motioneye page does not start. Can someone help me? thanks.

starbasessd, 2021-04-25:
A number of users, including myself, have reported issues with proxmox, but no solutions have been found. Hopefully someone who has a better background will be able to help.

zgadgeter, 2021-04-25:
Hmm, oh dear. Thanks for this info. Let's hope so. thanks

lmilius, 2021-04-25:
I just installed motion eye using the provided docker compose example into an lxc container on proxmox today, seems to be working fine for me, even after several reboots of the container, I haven't experienced any issues with it not loading. I am running an unprivileged container with nesting enabled and apparmor set to unconfined so that docker runs properly

balbulator, 2025-04-20:
> Сьогодні я щойно встановив motion eye за допомогою наданого прикладу створення докерів у контейнер lxc на proxmox, здається, у мене все працює нормально, навіть після кількох перезавантажень контейнера, у мене не виникло жодних проблем із завантаженням. Я запускаю непривілейований контейнер із увімкненим вкладенням і параметром apparmor для необмеженого, щоб докер працював належним чином
Can you explain in detail how exactly you enable this? I've just started learning Proxmox and I'm not entirely clear on how exactly you launch the container.

MichaIng, 2025-04-21:
Check the systemd service:
```sh
journalctl -u motioneye
```

zaphod2, 2026-02-07:
Das selbe Problem.
```console
# journalctl -u motioneye
Feb 07 18:42:05 motioneye systemd[1]: Started motioneye.service - motionEye Server.
Feb 07 18:42:06 motioneye meyectl[128]: configure_logging cmd motioneye: False
Feb 07 18:42:06 motioneye meyectl[128]: configure logging to file: None
Feb 07 18:42:06 motioneye meyectl[128]:     INFO: Hallo! Dies ist ein MotionEye-Server 0.43.1
Feb 07 18:42:06 motioneye meyectl[128]: CRITICAL: pid directory "/cam" does not exist or is not writable
Feb 07 18:42:06 motioneye systemd[1]: motioneye.service: Main process exited, code=exited, status=255/EXCEPTION
Feb 07 18:42:06 motioneye systemd[1]: motioneye.service: Failed with result 'exit-code'.
```

zaphod2, 2026-02-07:
`# journalctl -u motioneye`
Feb 07 18:55:17 motioneye systemd[1]: Started motioneye.service - motionEye Server.
Feb 07 18:55:18 motioneye meyectl[126]: configure_logging cmd motioneye: False
Feb 07 18:55:18 motioneye meyectl[126]: configure logging to file: None
Feb 07 18:55:18 motioneye meyectl[126]:     INFO: Hallo! Dies ist ein MotionEye-Server 0.43.1
Feb 07 18:55:18 motioneye meyectl[126]: CRITICAL: log directory "/cam" does not exist or is not writable
Feb 07 18:55:18 motioneye systemd[1]: motioneye.service: Main process exited, code=exited, status=255/EXCEPTION
Feb 07 18:55:18 motioneye systemd[1]: motioneye.service: Failed with result 'exit-code'.