motioneye-project/motioneye #2112: Raspberry Pi 4 - setup in the server configuration / Issue:  can't view the front end or view using IP address

RD7765, 2021-04-25:
I'm trying to set up a raspberry pi 4 (4GB model) as a server on wifi to view/record footage from 2-4 Pi Zero W's, and eventually utilize OpenVPN to access the feed outside my LAN. I would also like to attach an external hard drive to the RPI4  to have footage from each camera sent to the server, and finally set up email notifications but for now one step at a time.

Currently, I have 2 Pi Zero's running on wifi, and can view those feeds using their IP addresses on the LAN. They have motioneyeos installed. The raspberry pi 4 has the current version of buster installed, and I'm trying to add motioneye onto the OS.

The the IP's for the 2 cameras and the server (raspberry Pi 4) are static.  I've followed the instructions for the server installation here: https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
https://github.com/ccrisan/motioneye/wiki/Installation

The problem happens when I get to the section of the installation process "Accessing the Frontend" where it says to use the device's IP (in this case my server, right?) in the following format: "http://my server's static IP:8765/". When I use my raspberry pi 4's IP address from the Pi's browser or other browsers on the LAN it won't load. I've also tried using the IP's for the camera's and have had no luck. I also noticed one of the libraries in step 3 was substituted when I tried to install it.  Instead of 'libz-dev' it installed 'zlib1g-dev'.  

I tried checking other posts/issues for this problem. Thanks in advance.

starbasessd, 2021-04-25:
Pi 4:
Do you have a monitor & keyboard to view the console?
Have you installed SSH and enabled it?
Are you running Ubuntu 20.xx or RaspberryPiOS version of Buster?
Are you running Desktop, Full Desktop, or Lite or the equivalent?
Enough questions for now.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

RD7765, 2021-04-26:
> Pi 4: Do you have a monitor & keyboard to view the console? Have you installed SSH and enabled it? Are you running Ubuntu 20.xx or RaspberryPiOS version of Buster? Are you running Desktop, Full Desktop, or Lite or the equivalent? Enough questions for now.
> […](#)

Yes, I'm using a keyboard and monitor to view console.  SSH is installed and enabled. Running Raspbian buster (version 10), and believe the OS is either the lite version or basic desktop version (2021-03-04). Hope that helps.

starbasessd, 2021-04-27:
Confirm the IP address with the command
ip a
If you are trying to access the WebGUI on the PI 4 from the Pi 4's web
browser, use http://locaalhost:8765
If you are accessing from a PC web browser, use the ip address reported
above in
http://[ip_address]:8765
Run report for
netstat -apt | grep 8765

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

djpostka, 2021-04-28:
Hello,

i have the same problem, i'm following the wiki instruction, no error during installation but not access to the webGUI. 
netstat return a blank result. i have the good ip of the raspberry (3b+).
htop does not return anything that matches motionye.

i don't have access to the desktop browser.

djpostka, 2021-04-28:
ok, a find the problem, i have python 3.5 has default. The installation need to use pip2 and python 2.7. Now it's works

RD7765, 2021-05-02:
> Confirm the IP address with the command ip a If you are trying to access the WebGUI on the PI 4 from the Pi 4's web browser, use http://locaalhost:8765 If you are accessing from a PC web browser, use the ip address reported above in http://[ip_address]:8765 Run report for netstat -apt | grep 8765
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

I couldn't get it to work so I started over with a clean install on Pi Lite on the RP4, and now I can view the frontend using a browser on another computer on the network.  The cameras appear, and look to be working.  Now I just need to mount an external 1-2 TB external HD via USB to the RPI4.  Setting that up shouldn't be a challenge hopefully.

Thanks for your assistance!

closed 2021-05-02 by RD7765