motioneye-project/motioneye #2114: timelapse multiple days/timespan possible?

ozett, 2021-04-29:
If you installed a fresh version today, how would you have had multiple days snapshots to create the timelapse?
Please open a new issue, with at least the Base OS you are using, which install instructions you are using, which version of motion & motionEye were installed, what path is being used for File Storage, and what File Name is being used for Still Images

_Originally posted by @starbasessd in https://github.com/ccrisan/motioneye/issues/1022#issuecomment-829147226_

ozett, 2021-04-29:
you got me!
i installed yesterday, but after years still find no gui-button to create timelapse over time-periods, like day-overlapping hours or other day-periods. may i overlook something?

[image]
[image]
[image]
[image]

used: ubuntu 2004, installed word-by-word [like wiki tells](https://github.com/ccrisan/motioneye/wiki/%28Install-On-Ubuntu-%2820.04-or-Newer%29) (all as root). dedicated for motioneye, nothing else installed.

filenames:
``` 
-rw-r--r-- 1 root root  99900 Apr 29 00:00 00-00-00.jpg
-rw-r--r-- 1 root root  99827 Apr 29 00:20 00-20-00.jpg
-rw-r--r-- 1 root root  99915 Apr 29 00:40 00-40-00.jpg
``` 
some more information needed?

thanks.

starbasessd, 2021-04-29:
"what path is being used for File Storage, and what File Name is being used for Still Images"
Go to Settings
Go to File Storage
What path do you have for Storage Device
What path do you have for Root Directory?
Go to Still Images
What do you have for File Name?

As to creating a timelapse across multiple days or for a specific time period would need to have the paths and filenames standardized to make a search possible, instead of the flexibility currently available. There are a number of utilities and scripts available to create time lapses from .jpg files 
https://www.google.com/search?client=firefox-b-1-d&q=create+time+lapse+video+from+jpeg+files 
that you can modify to meet your needs
I have flagged as Enhancement & Feature

ozett, 2021-04-29:
Lighningfast reply. Wow! 👏 
okay, thanks for links and help. i will have a look.
and thanks for flagging enhancement. i will stay tuned .. 😉 

[image]

ozett, 2021-12-31:
found by accident another solution:
change naming of the files, to avoid subfolder
https://github.com/ccrisan/motioneyeos/issues/2261

>By default the name pattern of the still images is "%Y-%m-%d/%H-%M-%S" with the "/" it's creating the subfolders per-day. If I change the naming pattern so all the still images from multiple days end up in the same folder, then the "Timelapse, Group/folder" selection will find all the images from the multi-day collection of still images. For example : PrintName/%Y-%m-%d_%H-%M-%S