motioneye-project/motioneye #2115: Feature request: stream sequentially from different cams

stamanker, 2021-04-30:
Hi,

It would be nice if there are such feature exist that by specific port streams data X1 seconds from cam1, then X2 seconds from cam2, etc...

starbasessd, 2021-04-30:
Can you describe your request better? 
Are you asking to show full screen for X seconds per camera?
Are you asking to ONLY stream 1 camera at a time to the web interface (port 8081, whatever)?

starbasessd, 2021-04-30:
BTW, there are several add-ons for chrome browser, and if you want a Pi for displaying here's a great little setup (in my kitchen right now using a Pi3B+, 7" RPi Official Touch Display and case, that works well:
https://pimylifeup.com/raspberry-pi-kiosk/

starbasessd, 2021-04-30:
I just have it open each camera stream from my hub (5 camera streams) using the http://ip_address:8081 - 8085 urls, and cycles 5 seconds per camera. Great for keeping an eye on grandkids around the house....

stamanker, 2021-04-30:
For now i can configure cam1 on 8081, cam2 on 8082. And that is static binding. 
I'd like to have streaming on some special port, for example 8085, where i would get rotating: 3 seconds from cam1, then 1 second from cam 2, then 2 seconds from cam3...

JamBalaya56562, 2026-11-15:
A rotating stream would need motionEye to merge several camera streams into one, which it doesn't do. A browser or kiosk page that cycles through the camera stream URLs, as @starbasessd described, gives the same result. Cycling the cameras in motionEye's own UI is #455. Closing as not planned.

closed 2026-11-15 by JamBalaya56562