motioneye-project/motioneye #2117: Impossible to run motion eye

diazovitch, 2021-05-02:
Hi,

I install motion eye on my raspberry which contain domoticz and i can't run motion eye.
What is the problem please ?

```
pi@domoticz:~ $ sudo systemctl status motioneye
● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2021-05-02 19:10:00 CEST; 6min ago
  Process: 5326 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=exited, status=1/FAILURE)
 Main PID: 5326 (code=exited, status=1/FAILURE)

mai 02 19:09:59 domoticz systemd[1]: Started motionEye Server.
mai 02 19:10:00 domoticz meyectl[5326]: Traceback (most recent call last):
mai 02 19:10:00 domoticz meyectl[5326]:   File "/usr/local/bin/meyectl", line 5, in <module>
mai 02 19:10:00 domoticz meyectl[5326]:     from motioneye.meyectl import main
mai 02 19:10:00 domoticz meyectl[5326]:   File "/usr/local/lib/python3.7/dist-packages/motioneye/meyectl.py", line 28, in <module>
mai 02 19:10:00 domoticz meyectl[5326]:     import settings
mai 02 19:10:00 domoticz meyectl[5326]: ModuleNotFoundError: No module named 'settings'
mai 02 19:10:00 domoticz systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
mai 02 19:10:00 domoticz systemd[1]: motioneye.service: Failed with result 'exit-code'.

```

starbasessd, 2021-05-02:
Need to run keyed to with python 2, not python 3. Either change the default
or the startup command.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-05-02:
Sorry, auto correct in email program...
Need to run meyectl with python 2, not python 3. Either change the default
or the startup command.

diazovitch, 2021-05-02:
Sorry I’m noob. 
what should i do as a request?

starbasessd, 2021-05-02:
What is your base OS? Raspberry Pi OS? or something else?

diazovitch, 2021-05-02:
I'm on raspbian buster on rpi3b+

starbasessd, 2021-05-02:
Give me a bit, I'll get you the change.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-05-02:
Which install instructions did you use for motioneye?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

diazovitch, 2021-05-02:
This tuto : https://trevilly.com/installation-de-motioneye-sur-raspbian-buster-videosurveillance/

starbasessd, 2021-05-02:
Re-install over the top of the existing by using the instructions here:
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
then re-boot and connect to the motioneye at http://ip_address_of_the_pi:8765
and let me know.

diazovitch, 2021-05-02:
When i do this command : sudo wget https://github.com/Motion-Project/motion/releases/download/release-4.2.2/pi_buster_motion_4.2.2-1_armhf.deb dpkg -i pi_buster_motion_4.2.2-1_armhf.deb

I have symbol !?!

[image]

starbasessd, 2021-05-02:
Did you change to SuperUser by running sudo su first?
Did you do a copy/paste? If you do, you need to do each line separately.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

diazovitch, 2021-05-02:
Yes, i do each line separately ! It's ok now.
But when i do this command :

`sudo pip install motioneye`

I have this message below. Is it serious doctor ?

[image]

starbasessd, 2021-05-02:
Try this:
sudo pip2 install motioneye
I just built a RaspberryPiOS (buster), fully updated, and ran the instructions for installing from the github wiki, and it started up normally and I did not get that warning.

diazovitch, 2021-05-02:
It's the same thing.

[image]

starbasessd, 2021-05-02:
Now, when I run the command:
python --version
I get:    Python 2.7.15

diazovitch, 2021-05-02:
My python version is 3.7.3

I need this version for domoticz.

starbasessd, 2021-05-02:
Run the following commands:
ls -la /usr/bin/python*

diazovitch, 2021-05-02:
`~ $ ls -la /usr/bin/python*
lrwxrwxrwx 1 root root      24 août  30  2020 /usr/bin/python -> /etc/alternatives/python
lrwxrwxrwx 1 root root       9 mars   4  2019 /usr/bin/python2 -> python2.7
-rwxr-xr-x 1 root root 2984816 oct.  11  2019 /usr/bin/python2.7
lrwxrwxrwx 1 root root      36 oct.  11  2019 /usr/bin/python2.7-config -> arm-linux-gnueabihf-python2.7-config
lrwxrwxrwx 1 root root      16 mars   4  2019 /usr/bin/python2-config -> python2.7-config
lrwxrwxrwx 1 root root       9 mars  26  2019 /usr/bin/python3 -> python3.7
-rwxr-xr-x 2 root root 4275580 janv. 22 21:04 /usr/bin/python3.7
lrwxrwxrwx 1 root root      36 janv. 22 21:04 /usr/bin/python3.7-config -> arm-linux-gnueabihf-python3.7-config
-rwxr-xr-x 2 root root 4275580 janv. 22 21:04 /usr/bin/python3.7m
lrwxrwxrwx 1 root root      37 janv. 22 21:04 /usr/bin/python3.7m-config -> arm-linux-gnueabihf-python3.7m-config
lrwxrwxrwx 1 root root      16 mars  26  2019 /usr/bin/python3-config -> python3.7-config
lrwxrwxrwx 1 root root      10 mars  26  2019 /usr/bin/python3m -> python3.7m
lrwxrwxrwx 1 root root      17 mars  26  2019 /usr/bin/python3m-config -> python3.7m-config
lrwxrwxrwx 1 root root      14 mars   4  2019 /usr/bin/python-config -> python2-config
`

starbasessd, 2021-05-02:
Run:
sudo nano /usr/local/bin/meyectl

change line:
#!/usr/bin/python
to 
#!/usr/bin/python2
[ctrl] x  y [enter]
then run the commands
sudo systemctl start motioneye
sudo systemctl status motioneye

diazovitch, 2021-05-02:
In this file, there had : 

#!/usr/bin/python3
and i change to
#!/usr/bin/python2
but same results : 

[image]

starbasessd, 2021-05-02:
It's saying the motioneye.meyectl module didn't install. 
Run the following command:
sudo pip2 --version

diazovitch, 2021-05-02:
[image]

starbasessd, 2021-05-02:
Ok, pip2 isn't installed and it's using pip3 instead.
Run the following:
sudo curl https://bootstrap.pypa.io/pip/2.7/get-pip.py --output get-pip.py
sudo python2 get-pip.py
then run:
sudo pip2 --version

diazovitch, 2021-05-02:
`pip 20.3.4 from /usr/local/lib/python2.7/dist-packages/pip (python 2.7)
`

starbasessd, 2021-05-02:
run the commands
sudo systemctl start motioneye
sudo systemctl status motioneye

diazovitch, 2021-05-02:
Always the same error.
Maybe i have to reinstall ?

starbasessd, 2021-05-02:
OOps, may have forgotten:
sudo pip2 install motioneye

diazovitch, 2021-05-02:
I have this : 
[image]

starbasessd, 2021-05-02:
Ignore the update message

diazovitch, 2021-05-02:
You win 🥇 !

starbasessd, 2021-05-02:
Blame domoticz, it doesn't play nicely with apps needing python2.7

diazovitch, 2021-05-02:
It is not planned to provide an motion eye that works in python3 ? It's the future, no ?

starbasessd, 2021-05-02:
Yes, it's coming, but there are quite a few apps that were written in 2.7 that are still used. If you didn't notice, although they (Python 2.7 authors) wanted to end support back in 2008, announcing a sunset date of 2015. 6 years after 'sunset' they finally End Of Life'd it, no more updates. 2008 to 2021. 13 years, and people are still using python 2.7. Better than Windows.

starbasessd, 2021-05-02:
Probably should either hope for better backward compatibility for 4.0, or start building for 4.0 [grinning]

diazovitch, 2021-05-02:
Thank you for taking the time to resolve my issue. I could never have done it alone. Thank you from France.

starbasessd, 2021-05-02:
If you are happy, please close.

closed 2021-05-02 by diazovitch