motioneye-project/motioneye #2118: Why button action doesn't appear ?

diazovitch, 2021-05-05:
Hi,

I create 4 files in /etc/motioneye :
up_6
down_6
right_6
left_6
My camera id is "6" and i would like to use the pan tilt fonction.
It is a camera HEDEN.

I restart motioneye.

What is the problem ?

starbasessd, 2021-05-05:
the file up_6 needs to be a script or executable that talks to the camera
and tells it to move up.
each camera brand does it differently, and some can't be done at all that
way.
I do not have specific examples for yours, and I know one of mine can only
be done by their custom app.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

diazovitch, 2021-05-05:
Ok I understand that but the button should appear even if the file was empty?

starbasessd, 2021-05-05:
What are the file permissions? Are the files set to executable?
How is motionEye installed?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

diazovitch, 2021-05-05:
Nice ! It's a problem of chmod ! Thank you.
Do you know what i have to add in the file ?
I have the html link for PTZ option but how have to write ?

starbasessd, 2021-05-05:
I created a file 'up_1' in /etc/motioneye by using 'sudo touch up_1' (0
length blank file)
I then reloaded my motionEye WebGUI (F5)
Nothing showed.
I then ran sudo chmod +x up_1 (makes it executable)
Reloaded WebGUI (F5)
Now shows up arrow next to snapshot icon in lower right of camera 1 menu
screen

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-05-05:
No, I don't. Sometimes you can use 'curl', or 'wget/wput' or some other
variation, but it's beyond the scope of my support at this  time. Someone
else may have ideas. CCRISAN has written a very flexible tool, but I don't
have a clue on that..

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

diazovitch, 2021-05-05:
I find on the web for my camera Heden :

IP ADRESS + PORT : 192.0.0.XX:XX 
URL JPG: snapshot.cgi
URL flux MPEG: video.cgi
URL move to left : decoder_control.cgi?command=4&onestep=7
URL move to right : decoder_control.cgi?command=6&onestep=5
URL move up : decoder_control.cgi?command=0&onestep=1
URL move down : decoder_control.cgi?command=2&onestep=3
URL stop move : decoder_control.cgi?command=5

closed 2021-05-05 by diazovitch