motioneye-project/motioneye #2119: Motioneye on Linux Mint 20

StefanGose91, 2021-05-06:
I am on the following Linux Mint 20 Cinnamon:
nutzer@machine:~$ uname -r
5.4.0-26-generic
nutzer@machine:~$ uname -v
#30-Ubuntu SMP 

I cant find the following packages threw apt, thats why i dont get pip2 to work. 
E: Paket python-pip kann nicht gefunden werden.
E: Paket python-tornado kann nicht gefunden werden.
E: Paket python-pip kann nicht gefunden werden.

Is there a possible solutions?
Thanks in advance
Stefan

starbasessd, 2021-05-06:
Try following the instructions for 
https://github.com/ccrisan/motioneye/wiki/%28Install-On-Ubuntu-%2820.04-or-Newer%29
If they don't work, let me know and I'll research further.

StefanGose91, 2021-05-06:
Thank you very much, installing pip2 with python worked fine :)

closed 2021-05-06 by StefanGose91