motioneye-project/motioneye #2120: Motioneye isn't compatible with Reolink E1 pro ?

diazovitch, 2021-05-06:
I have 6 cameras on motioneye :

- 1 Picam
- 1 Reolink RLC-420
- 1 Reolink RLC-520
- 1 Foscam FI9900P
- 1 Foscam C1
- 1 HEDEN

I try to associate my last camera : REOLINK E1 PRO but it doesn't work.

I use: 
- rstp://192.168.X.XX:554/h264Preview_01_sub
- rstp://192.168.X.XX:554/h264Preview_01_main
- rstp://192.168.X.XX/h264Preview_01_main
- rstp://192.168.X.XX/h264Preview_01_sub
- rstp://user:password@192.168.X.XX:554/h264Preview_01_sub...

None works !
Do you have a solution please ?

cap9qd, 2021-05-06:
Have you looked at this article:
https://support.reolink.com/hc/en-us/articles/360007010473-How-to-Live-View-Reolink-Cameras-via-VLC-Media-Player

And note the comment about 8mp cameras being h265 and not h264.

I also had better luck with the rtmp streams from my ReoLink cameras but
others experience may vary.

diazovitch, 2021-05-06:
The reolink E1 pro is a 4 MP camera !

On your link we can see :+1: 
_"Main Stream: rtsp://admin:password@ip_address:554//h264Preview_01_main"_

But my 2 reolink working, i use this url:

RLC-520: rtsp://192.168.X.XX**/**h264Preview_01_sub **(without port and whit just 1 "/" between IP and h264)**

RLC-420: rtsp://192.168.X.XX**/**h264Preview_01_main **(without port and whit just 1 "/" between IP and h264)**

starbasessd, 2021-05-06:
Try
rstp://192.168.X.XX/h264Preview_01_sub
without the port #. On the iSpy page 
https://www.ispyconnect.com/camera/reolink
it doesn't create the URL with the port number included.

diazovitch, 2021-05-06:
I think it is not possible to connect the camera.
It is very strange that I cannot set the port.

starbasessd, 2021-05-06:
Check the rest of the suggestions on the iSpy page, and check if your firmware can be upgraded on the camera. Several other uses have reported that a camera might not work with the original firmware, but works after the firmware is updated.

starbasessd, 2021-05-06:
https://support.reolink.com/hc/en-us/articles/900004231023-28-01-2021-Firmware-for-E1-Pro-513SD5-

starbasessd, 2021-05-06:
https://support.reolink.com/hc/en-us/sections/360002374874-Firmware

diazovitch, 2021-05-06:
I update the firmware but it the same thing.

diazovitch, 2021-05-06:
I try with VLC without success.

diazovitch, 2021-05-06:
I do an error. It's a Reolink E1 not PRO. I change the title of the post. And firmware has been update.

starbasessd, 2021-05-06:
Sounds like something is broken.

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-05-06:
E1 doesn't show for iSpy, only E1 Zoom, and E1 Pro. May not have local web support, only App support.

diazovitch, 2021-05-06:
Oh !!
I have read too quickly !
I return this camera to amazon and i buy a new one but PRO ! Thank you for your help.

starbasessd, 2021-05-06:
Don't forget to close this

closed 2021-05-06 by diazovitch