motioneye-project/motioneye #2121: use a wifi extender connection issue

AlgeaPool, 2021-05-16:
i have a few cameras up and running MotionEye and the whole system works great. Amazing softs ccrison. i introduced a wifi extender into the mix and have lost connection to the camera that is now using the wifi extender. The IP address of the Camera is the same as before but the MotionEye OS is not finding the previously connected camera. Ofc the IP of the camera is now connected through the IP address of the wifi extender. Thus, ...... (guessing now) i need to instal Motioneye on the Wifi Extender? or better yet.... create a 'pass through'. Any help would be appreciated. Thanks

starbasessd, 2021-05-16:
Are you using motionEye or motionEyeOS? Which Version(s)?
What hardware? PiZero, 1, 2b, 3b, 3b+, 4b 1GB, 2GB, 4GB, or 8GB?
Can you ping the camera from a PC on the same network?
Can you ping the PC from the motionEye?
Which WiFi extender?

starbasessd, 2021-05-16:
Pass through is the way you want to go. Allow all traffic both directions. No firewall. It needs to act as a 'switch', not a 'router'.

AlgeaPool, 2021-05-16:
pretty sure it is a MotionEye install for Raspbian. from this page
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
version ? (connection issue at the moment), installed and updated 1 to 2 months ago.

Raspberry Pi 3b (possibly a 3b+, not sure) , rpi Camera v2.1.

no other softs or purposes for this Pi. (only use SSH connection) 

Ping form PC on Network.......

C:\Users\shann>ping 192.168.2.238

Pinging 192.168.2.238 with 32 bytes of data:
Reply from 192.168.2.215: Destination host unreachable.
Reply from 192.168.2.215: Destination host unreachable.
Reply from 192.168.2.215: Destination host unreachable.
Reply from 192.168.2.215: Destination host unreachable.

Ping statistics for 192.168.2.238: 
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),

Using Putty/SSH returns 'Network connection error. timed out'). 

Note. My RPI3b has become unreachable via SSH now that i logged it into the Wifi extender....(after the the SSID was changed to Wifi Extenders). 

Note. connecting to the Router through Browser confirms the Wifi Extender is connected (with a strong signal) and the RPI3b is connected BUT has a 'POD' connection.

_Can you ping the PC from the _motionEye?__    not sure how to do this.

i am running the typical setup.....5 cameras on; a Zero w, and couple Pi2b, THIS pi3b and a 
Pi4 2g. Feeding into a Master/Server pi3b running MotionEye which ofc shows all 5 (now four) Cameras on the same Browser page. All accessible from a PC.

The Wifi Extender is a Joowin 300mbps Wifi Range Extender model: JW-WR3025

AlgeaPool, 2021-05-16:
_Pass through is the way you want to go. Allow all traffic both directions. No firewall. It needs to act as a 'switch', not a 'router'._

huh?  'It' is the Wifi Extender? i will look into config options for the Wifi Extender?? Only 'firewall' i know is on the 
PC

starbasessd, 2021-05-16:
Can you connect a laptop, tablet or phone to the Extender? 
Then try to connect to the Pi. 
Let me know.

starbasessd, 2021-05-16:
Do you have a monitor and keyboard you can connect to the Pi?

AlgeaPool, 2021-05-17:
i have found the Config (via Browser) page for the Joowin Wifi extender. Seems i have options...... 

[image]

Appears to be in Router Mode. I'm thinking (limited, sorry) i might need to engage the Repeater Mode except unsure about that exactly....any help appreciated, thanks

[image]

AlgeaPool, 2021-05-17:
[image]
 This photo is from the Router Mode setup page. There is a Drop down list for WAN Access Point which gives options: Static, DHCP or PPPoe. PPPoe appears selected. DHCP was its default setting.

starbasessd, 2021-05-17:
You want either AP or Repeater, and if they can turn off DHCP, do so.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

AlgeaPool, 2021-05-18:
I'm pretty much in Repeater Mode....AP Mode, which is Access Point seems to be for wired connection and appears to kill wireless connections...Repeater Mode has limited options

[image]
[image]

and this is what the Advanced cog, up in the corner, looks like
[image]

Currently, i am unable to SSH or MotionEye into/connect to the Motioneye Camera 238 that is connected to the WAP-WifiAccessPoint Joowin Device. 

Note. The Joowin Device serves up good LAN Wifi for phones and PC, connect and surf, signed into Joowin. 

same as the above post......Win PC, cmd returns

Reply from 192.168.2.215: Destination host unreachable.
Ping statistics for 192.168.2.238:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),

the 'Received = 4' telling me the Camera238s is connected some how into the Lan throug the Joowin Wifi extender.

starbasessd, 2021-05-18:
You need to contact Joowin support, ask them how to set up the repeater so it will pass dhcp requests through from the Pi to your modem/router and back again, so the Pi gets an IP address on your network.

AlgeaPool, 2021-05-22:
They got back to pretty quickly after i found the email address at the bottom to page. The Info is such:

You can setup AP mode
Please note: AP mode need to change Automatic IP to manual IP, change IP address to  "192.168.10.2". After setup, you also need to change Manual IP to Automatic IP again.
The setup steps of AP mode as follows:
1. Use the network cable connect your router and the extender.
2. Plug in the extender, then the lights will all light up. Wait about 60s, connect WiFi “JOOWIN-XXX" with your computer. 
3. Open Network and Sharing Center. Click Internet, and then click the “attribute”. Choose "TCP/IPV4", and then change Automatic IP to manual IP, change IP address  to  "192.168.10.2". Enter "255.255.255.0" in Subnet mask.
4.Then please login IP: 192.168.10.1 on Google Chrome,  enter the username and password: admin, then you can set up it.
5. Select "AP Mode" above. Change the SSID (new wifi name) and Password (you can set a new password different from your router). Click Apply.
6. After the setup is complete, it will reboot automatically. Then you need to change Manual IP to Automatic IP  again, and then connect the new wifi.

it'll be a few more days before i can try it out, will update soon

AlgeaPool, 2021-05-25:
After a little back and forth the following appears to be the Solution....

After setting up Repeater Mode and device reboot. 

Close the security mode: go to the configuration page 192.168.10.1, then select "System", select "Security Mode Setting" and select "Apply" in "Disable Security Mode".

This appears to have resolved the issue. Although IP address changed it was rediscover able.

closed 2021-05-25 by AlgeaPool