motioneye-project/motioneye #2122: Unable to load stream ERROR: mjpg client timed out receiving data for camera 1 on port 8081,

mwy6283, 2021-05-17:
I have scoured the web and the forum for months but i am not able to find unable to find a solution. Cameras Reolink 410 5mp which are recognised in HA integration and  VCL.  Running the addon thru docker on a RPI4 4g. The cameras are recognised in HA and VCL.

 Any advise and help would be appreciated

WARNING: 403 GET /login/?_=1620883446234&_username=_&_signature=1f4403faf8d6a2a53894d43e0b4922fad0930dea (192.168.0.38) 1.37ms,
   ERROR: mjpg client timed out receiving data for camera 1 on port 8081,
   ERROR: mjpg client timed out receiving data for camera 1 on port 8081,
   ERROR: mjpg client timed out receiving data for camera 1 on port 8081,
   ERROR: mjpg client timed out receiving data for camera 1 on port 8081,
   ERROR: mjpg client timed out receiving data for camera 1 on port 8081,
   ERROR: mjpg client timed out receiving data for camera 1 on port 8081,
    INFO: removing camera config file /etc/motioneye/camera-1.conf...,
    INFO: hello! this is motionEye server 0.42.1,
    INFO: cleanup started,
    INFO: wsswitch started,
    INFO: tasks started,
    INFO: mjpg client garbage collector started,
    INFO: server started,
Failed to open /dev/video0: No such file or directory,

starbasessd, 2021-05-17:
Do you have a Pi to test on? If so:
Install RaspberryPiOS
install Docker (I use apt install docker.io, but have tested with docker-ce)
Use docker run command from wiki:
docker run --name="motioneye" \
    -p 8765:8765 \
    --hostname="motioneye" \
    -v /etc/localtime:/etc/localtime:ro \
    -v /etc/motioneye:/etc/motioneye \
    -v /var/lib/motioneye:/var/lib/motioneye \
    --restart="always" \
    --detach=true \
    ccrisan/motioneye:master-armhf
Add reolink camera with:
Network Camera
URL: rtsp://IP_address/h264Preview_01_sub
Try both TCP and UDP
If it works there (It Should, I tested there) then the issue lies with the HA wrappers.

mwy6283, 2021-05-17:
Hi I used a  the same command  as the wiki when i installed the current  addon , i am now getting the following error when trying to load the  as above.

`pi@raspberrypi:/volume1/docker $ sudo docker run --name="motioneye3" -p 8765:8765 --hostname="motioneye" -v /etc/localtime:/etc/localtime:ro -v /etc/motioneye:/etc/motioneye -v /var/lib/motioneye:/var/lib/motioneye --restart="always" --detach=true ccrisan/motioneye:master-armhf
WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm/v7) and no specific platform was requested
1d96488371b42e9d58d201a01c98bd9e9efe9f033d48d51cc447900c67d02dc9. Not sure what the problem is ?

starbasessd, 2021-05-17:
If running on a Pi as the instructions state, you need to change the very
last bit from

ccrisan/motioneye:master-amd64

to
ccrisan/motioneye:master-armhf

The error indicates you used the amd64 line

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mwy6283, 2021-05-17:
Hi  I am confused . I have  installed using ccrisan/motioneye:master-armhf s shown in my previous post.  why is it  saying that i should install  arm/v7 .
Sorry to take  up your time on this but as i have mentioned ai have tried many option  installing  thru HA addon same result Grey screen which  indicates that fps is increasing  but no image
[image]
[image]

starbasessd, 2021-05-17:
Are you running on a Pi? Or a PC?

starbasessd, 2021-05-17:
It could also be you have too many conflicting packages installed. I'd try me previous suggestion, or just HA and motionEye, and get them to work, then move to other packages. You may be asking your system to do too much, also.

mwy6283, 2021-05-17:
I have just restarted Motioneyes with armhf here is the log ?
  INFO: hello! this is motionEye server 0.42.1

    INFO: cleanup started

    INFO: wsswitch started

    INFO: tasks started

    INFO: mjpg client garbage collector started

    INFO: server started

   ERROR: mjpg client timed out receiving data for camera 1 on port 8081

   ERROR: mjpg client timed out receiving data for camera 1 on port 8081

   ERROR: mjpg client timed out receiving data for camera 1 on port 8081

   ERROR: mjpg client timed out receiving data for camera 1 on port 8081

   ERROR: mjpg client timed out receiving data for camera 1 on port 8081

   ERROR: mjpg client timed out receiving data for camera 1 on port 8081

   ERROR: mjpg client timed out receiving data for camera 1 on port 8081

   ERROR: mjpg client timed out receiving data for camera 1 on port 8081

   ERROR: mjpg client timed out receiving data for camera 1 on port 8081

starbasessd, 2021-05-17:
I would delete and re-add the camera

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mwy6283, 2021-05-17:
I  am running RPI4 4G  with OMV5 & docker. I mentioned that  i had been running with HA on a Imac PC but this is not running at the moment  but i had the same issue . So i decided to  install everything on the PI.  all off my cameras are recognised in HA Lovelace  with the rtsp stream 

 I will delete the camera and see what happens. Should I delete the camera from the Reolink app do you think this would work?

starbasessd, 2021-05-17:
No, just the Pi/MotionEye

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mwy6283, 2021-05-17:
I did not install Motioneye OS on the Pi , i only installed  the docker Motioneye image from Docker. What i cannot understand is why i was unable to use Motioneye  from the addon in HA on my imac via VM as i had the exactly the same problems. according to the videos etc it works out of the box ?

starbasessd, 2021-05-17:
I wouldn't have a clue, as FRENCk created the wrapper for HA. It may be a
HA problem, it may be a HA motionEye wrapper issue, but if you run the
docker (I regularly build machines (Pis and VMs with various flavors OS)),
and they
 all work, without HA. I cannot get HA based machines to work with FRENCK's
motioneye plugin. I don't know what to tell you. THere are others that have
posted similar issues, and Freck says I should be able to help, but that
hasn't been the case. If you take the plugin and HA out of the picture,
motionEye works in Docker or as a direct install.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mwy6283, 2021-05-18:
Ok i installed the script./writeimage.sh -d /dev/mmcblk0 -i "/path/to/motioneyeos.img"  (20200606) from wiki  directly onto RPI4. went to Ip address  and still end up with nothing. is this script only for RPI cameras I can always get to this point in any application but no image. see  snapshot
[image] 
I keep on wondering if it is my iMac so I may try windows application later as nearly every video they use windows?? it so frustration been trying now for over a month. I am not interested in the HA motioneye but would l love to be abale to set it up in a docker container on its own.

 thanks for your help.

starbasessd, 2021-05-18:
You changed the Surveillance Username to 'admin'. The Admin username is 'admin', so you are confusing the system. It is only giving the admin account the viewer rights.  If you want a different username for Surveillance, try 'camera' or 'operator'. They cannot be named the same.

starbasessd, 2021-05-18:
My iMac is from 2009, and the docker installs I've tried so far require iMacs from 2010 or later. 
I'll look into it further, and see what I can find. I have installed docker.io and docker-ce in Raspberry Pi 4s and it works well there. I've run into issues (that others brought here) with any kind of 'portainer' type management supervisors on any flavor of linux. 
I just thought of something of a work-around for my iMac, I'll report later.

mwy6283, 2021-05-18:
I have checked with Reolink and the username = admin cannot be changed only the password. See below. 

Step 1: Launch Reolink Client and log in to your device. Step 2: Click Device Settings>Manage User, then select the user and click Modify Password button. Notes： The admin password can be modified while the name-admin can not be changed.

 I wonder how others have solved this particular problem ?

starbasessd, 2021-05-18:
The section where you are changing Surveillance User username is for MotionEye, not for your camera. You have 2 users for motionEye, 'admin' which has access to all settings, and Surveillance User 'user' by default, that can only watch the cameras, not make changes.
The Reolink admin has nothing to do with motionEye, only when you add the camera to motionEye.

330chauf, 2021-07-30:
Hi There, I know this is a bit back dated but I was just wondering if you got the grey screen issue sorted.

I have 4 reolink camera’s running on SYNOLOGY NAS and SSS. All four work fine. If I connect them to HA through the SYN integration all work fine. But when pickup the feed from the camera’s in motioneye I, like you get a grey screen.

I am intrested to hear on if and how you got sorted?

KR

Edit: Small update, One of my Camera’s is working. YES. However when adding a second camera with the same FW and identical setting I get the message in Motioneye “UNABLE TO OPEN VIDEO DEVICE”

It’s a Mystery to Me :wink:

lmai95, 2022-10-23:
Hi guys and gals,
I have the same issue but in a much more simple system, no docker or anything. Just one Raspberry Pi B 3+ with the motionEyeOS 20200606 installed on it.[image]

I'm  using this [camera](https://electreeks.de/shop/raspberry-pi-kamera-nachtsicht-mit-infrarot-sperrfilter-175-grad-weitwinkel-objektiv/). The file aren't uploaded any where, just reviewed via the android app or the web browser. Realy just the out of the  box experience. 

I had the same problem with  a  Raspberry Pi B 3 and an old camera  but supected the hardware so I replaced it and it didn't change. 

Here  are my logfiles:
[motion(2).log](https://github.com/motioneye-project/motioneye/files/9846062/motion.2.log)
[boot.log](https://github.com/motioneye-project/motioneye/files/9846063/boot.log)
[dmesg.log](https://github.com/motioneye-project/motioneye/files/9846064/dmesg.log)
[motioneye.log](https://github.com/motioneye-project/motioneye/files/9846065/motioneye.log)

lmai95, 2022-11-03:
I've found a report concerning this topic [Forum Link](https://bleepcoder.com/spksrc/386804100/camera-stops-working-several-times-a-day)

>just FYI: I installed yesterday a brand new Raspbian Stretch onto a USB stick which is booted by a raspberry Pi 3B+ >running the latest motion 4.2.1 and the latest MotionEye. Until now everything OK, motion is detected and recorded >without interruptions or losing connection to the connected raspberry pi cam.
>With other words topic problems are located in MotionEyeOS image.
>Since MotionEye manually installed on Raspbian runs fine (and motion, ... packages are upgradable) I'll abandon the >MotionEyeOS path for now. MotionEyeOS also can't boot from USB. So Raspbian installation is the way to go for me.

jluther23, 2022-12-09:
> You changed the Surveillance Username to 'admin'. The Admin username is 'admin', so you are confusing the system. It is only giving the admin account the viewer rights. If you want a different username for Surveillance, try 'camera' or 'operator'. They cannot be named the same.

This fixed my issue with the white screen.  Thank You!!!

JamBalaya56562, 2026-11-11:
Closing, as this was the add-on with motionEye 0.42.1, and the later replies are different setups (motionEyeOS, a Synology NAS). One cause found here: the Surveillance User must not have the same name as the admin, as jluther23 confirmed. Please open a new issue with `motion.log` if a Reolink camera still shows nothing with a current release.

closed 2026-11-11 by JamBalaya56562