motioneye-project/motioneye #2123: Can't get Motioneye to run on top of Raspbian  buster 10 full desktop

ADBW007, 2021-05-18:
HI,

I have followed the steps on the Wiki page for installing motioneye on  raspberry pi.  I have the latest  version of raspberry pi OS.  I have made sure that I have been using python 2 and also sudo at the front of the commands.  

I have tried this with installing motion first and without.  

If I have not installed motion first when I run the status command it tells me motion is not working.  To be clear the application I have asked to be enabled and started is Motioneye.

I'm lost as to what I am doing wrong as I am simplely blindly following the commands.  I am more of a hardware engineer and happy to help others if they have electronics problems.  My aim is to write a simple program to access the GPIO when there is motion detected so not only will it record but I will able to interact with a bit of electronics.  i.e switch  alight on then maybe a warning "get back you ba**tards,  I'll break your legs"  before maybe a 130dB at 1 metre sounder.  If successful will up load the program here.   

If someone has had some success with this please could the upload an image,  minus any personal setting i.e. wifi and things like that.  I am quite happy to use ethernet to then setup the wifi.

starbasessd, 2021-05-18:
Can you ssh or use a keyboard, mouse & monitor and run the following command:
sudo systemctl status motioneye
and post the results here, please?
Also run the commands:
python --version
pip --version
and post the results, please.

starbasessd, 2021-05-18:
I am building a new Pi3B+, RaspiOS 20210304 from RPIFoundation Downloads, full update, full desktop. 
Will be loading instructions from
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
Will post my results in a few...

starbasessd, 2021-05-18:
Confirmed default python and pip:
root@raspberrypi:/home/pi# python --version
Python 2.7.16
root@raspberrypi:/home/pi# pip --version
pip 18.1 from /usr/lib/python2.7/dist-packages/pip (python 2.7)
-----
sudo su
apt update && apt upgrade -y
apt-get install ffmpeg libmariadb3 libpq5 libmicrohttpd12
wget https://github.com/Motion-Project/motion/releases/download/release-4.2.2/pi_buster_motion_4.2.2-1_armhf.deb
dpkg -i pi_buster_motion_4.2.2-1_armhf.deb
apt-get install python-pip python-dev libssl-dev libcurl4-openssl-dev libjpeg-dev libz-dev
pip install motioneye
mkdir -p /etc/motioneye
cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf
mkdir -p /var/lib/motioneye
cp /usr/local/share/motioneye/extra/motioneye.systemd-unit-local /etc/systemd/system/motioneye.service
systemctl daemon-reload
systemctl enable motioneye
systemctl start motioneye
-----
systemctl status motioneye
● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset
   Active: active (running) since Tue 2021-05-18 13:26:23 EDT; 15s ago
 Main PID: 1685 (meyectl)
    Tasks: 3 (limit: 2062)
   CGroup: /system.slice/motioneye.service
           ├─1685 /usr/bin/python /usr/local/bin/meyectl startserver -c /etc/mot
           ├─1693 /bin/sh -c /usr/bin/ffmpeg -codecs -hide_banner
           └─1694 /usr/bin/ffmpeg -codecs -hide_banner

May 18 13:26:23 raspberrypi systemd[1]: Started motionEye Server.
May 18 13:26:34 raspberrypi meyectl[1685]:     INFO: hello! this is motionEye se
-----
Opened desktop browser in running raspberry pi:
http://localhost:8765
Log in as admin with no password
working
open browser on Win10 machine
http://pi_ip_address:8765
working

ADBW007, 2021-05-18:
I'm able to SSH,  I have mastered that part of headless setup. 

Raspberry Pi reference 2021-03-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 461109415073d2eb67083709662ba983cc191f14, stage5

Python 2.7.16
pip 18.1

● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2021-05-18 22:08:14 BST; 2min 50s ago
 Main PID: 371 (meyectl)
    Tasks: 5 (limit: 4915)
   CGroup: /system.slice/motioneye.service
           ├─371 /usr/bin/python /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
           └─556 /usr/bin/python /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf

May 18 22:08:14 raspberrypi systemd[1]: Started motionEye Server.
May 18 22:08:16 raspberrypi meyectl[371]:     INFO: hello! this is motionEye server 0.42.1
May 18 22:08:17 raspberrypi meyectl[371]:     INFO: main config file /etc/motioneye/motion.conf does not exist, using default values
May 18 22:08:17 raspberrypi meyectl[371]:     INFO: cleanup started
May 18 22:08:17 raspberrypi meyectl[371]:     INFO: wsswitch started
May 18 22:08:17 raspberrypi meyectl[371]:     INFO: tasks started
May 18 22:08:17 raspberrypi meyectl[371]:     INFO: mjpg client garbage collector started
May 18 22:08:17 raspberrypi meyectl[371]:     INFO: server started

starbasessd, 2021-05-18:
> MotionEye is running.
> Access either from http://localhost:8765 or http://pi_ip_address:8765
> Thanks
> Kevin Shumaker
> Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>
> N38° 19' 56.52"
> W85° 45' 8.56"
> Semper Gumby
> “Don't tell people how to do things. Tell them what to do and let them
> surprise you with their results.” - G.S. Patton, Gen. USA
> Ethics are what we do when no one else is looking.
> Quis custodiet ipsos custodes?
> “There is no end to the good you can do if you don’t care who
> gets the credit.” - C Powell
> You know we're sitting on four million pounds of fuel, one nuclear weapon
> and a thing that has 270,000 moving parts built by the lowest bidder. Makes
> you feel good, doesn't it?

I see you used pi-gen to build your own. What changes do you make? Do you
add aparmour, or firewalls or anything like that?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ADBW007, 2021-05-18:
Thank you, it is working,  I feel silly.  I just copied the image from the raspberrypi.org website and used there app to write the image.

starbasessd, 2021-05-18:
The difference between motionEye and motionEyeOS is motionEye WebGUI is on port 8765, and motionEyeOS is on the normal http port 80

closed 2021-05-18 by starbasessd