motioneye-project/motioneye #2124: emulate motion not working correctly

MrDomingo, 2021-05-19:
It appears that the camera selector is zero based but 0 as an index is invalid choice.
http://{IP}:7999/{camera}/config/set?emulate_motion={value}

I have 3 cameras, Camera1, Camera2, Camera3 configured.

Test for "camera #1"
http://{IP}:7999/1/config/set?emulate_motion=0
Response is:

<– back

Camera 1 -- Camera2
emulate_motion = 0
Done

Test for "camera #2"
http://{IP}:7999/2/config/set?emulate_motion=0
Response is:

<– back

Camera 2 -- Camera3
emulate_motion = 0
Done

**Test for "camera #3" ==> wraparound to Camera1**
http://{IP}:7999/3/config/set?emulate_motion=0
Response is:

<– back

Camera 3 -- Camera1
emulate_motion = 0
Done

If one tries "camera #0" no mapping occurs
<– back

Camera 0
emulate_motion = 0
Done

This can be seen more clearly with following tests:

http://{IP}:7999/0/detection/status
Camera 0 Detection status NOT RUNNING

http://{IP}:7999/1/detection/status
Camera 1 -- Camera2 Detection status ACTIVE

http://{IP}:7999/2/detection/status
Camera 2 -- Camera3 Detection status ACTIVE

http://{IP}:7999/3/detection/status
Camera 3 -- Camera1 Detection status ACTIVE

http://{IP}:7999/4/detection/status
The requested URL is not valid.

Clearly the camera selected is via zero based index and the software at least does a wrap around (mod 3) which allows me to control all 3 but this is clearly a bug.

Ned

starbasessd, 2021-05-19:
I've seen this when cameras are added, deleted, and re-added.

starbasessd, 2021-05-19:
The order issue in motion could be related to how motionEye makes the changes (adding and deleting valid lines) in motion.conf. I am not a good enough programmer to parse how it makes the changes. (don't even get me started in 'sed' or 'sql' (grin))...

MrDomingo, 2021-05-19:
I at least have IT background and could poke around to resolve it but most other people would give up on it.  I am experimenting on motion detection using a PIR sensor and a radar sensor and so was experimenting with motion trigger options.  Something like RPI4 can handle a couple of cameras with software motion detection but from experience using image analysis to detect motion is pointless, except in controlled indoor environments.  Even though that motion trigger is available, I still see a problem in that once recording is triggered, and switch off message is lost or not issued for some reason, the system will record forever.  Would have been nice to add a timeout option on "emulate_motion=1".

starbasessd, 2021-05-19:
motionEye does have a length limiter:
Settings, Movies, Maximum Length.
I have a feeling you are trying to program motion (Motion-Project) directly.
Take a look at the WiKi for motionEye. 
https://github.com/ccrisan/motioneye/wiki
Here is the motion (Motion-Project github, with a lot of specific resources
https://github.com/Motion-Project/motion
I use motionEye to monitor a couple of outdoor cameras (along with indoor cameras), and between masking and sensitivity, I seldom get false positives or negatives.

MrDomingo, 2021-05-19:
In my situation we have many trees around the house and with shadows and movement from wind it is hard to setup motion detection that does not trigger due to shadows.  When I restrict view via mask as much as I can I then get no triggers or triggers that are too late.  No point on triggering when someone is already at the door and their face is not visible.

starbasessd, 2021-05-20:
Again, are you trying to program motion directly?
There are a number of websites with info on GPIO and motioneye, some refering directly to PIR, some with outputting to GPIO, etc:
https://www.google.com/search?client=firefox-b-1-d&q=motioneye+gpio
When I know I am going to be extremely occupied, I have a motion trigger in motionEye play a sound through a script to act as a 'doorbell' for my driveway that works well, too. Instead of doorbell.mp3, it could easily play your message through an outside speaker, if I were so inclined.

MrDomingo, 2021-05-21:
I am accessing motion directly as I was unaware of any similar facility in motioneye.  Btw, I gave up on using RPI4 because it had issues in particular setup I chose.  I had it setup with bootable SSD drive + SATA to USB3 on one USB3 port and another HDD + SATA to USB3 adapter on second port.  HDD was mounted to /mnt/archive folder.  Occasionally the HDD disappeared and motioneye then filled the SSD with videos.  This problem was too frequent to make the RPI4 viable in this configuration. I am currently using a Intel NUC running Debian distro. If motioneye has alternative control mechanism I will look at using that.
I am software developer with close to 40 years in the field, but in areas more related to graphics, postscript rips and print devices, and finally in print industry.  My interest in electronics and microcomputers in mid-70's led me into IT but now that I am semi-retired I am again playing with hardware, i.e. MCUs like 8266 and ESP32.  I have a zmodo security camera system that is now a few years old and has issues.   Being unable to maintain it led me to work towards a home built solution that would be based on motioneye and some custom built PCBs to complement the motioneye setup.  My initial test setup has Dahua 4MP camera and into this I added two of the 6 zmodo cameras.  The NUC cpu has a 60% load with these 3 cameras doing motion detection.   My hope is to remove the need for software motion detection and rely on PIR + Radar sensors.  Once this works I will retire zmodo box and put spare Dahua camera and some more zmodo cameras under control of motioneye system.  PIR sensor I am using is a small outdoor sensor made by Panasonic, EKMC160111.  The Radar sensor is HFS-DC06. Where I live the PIR sensors will not work on hot days in summer so radar is a fallback.  My custom hardware is based on ESP32 MCU and would rely on WiFi for connectivity but my policy is to stay away from WiFi for cameras and anything else related.  Once concept is proven I will redesign sensor boards to be connected using ethernet + POE.

zagrim, 2021-05-21:
@MrDomingo The disappearing HDD might've been due to insufficient power and/or the RPi having trouble keeping voltage up in case the SATA to USB3 adapter didn't have external power supply attached. I have no experience on RPi4 but tradidionally RPis have not been too great at supplying power. I have a RPi3 with an SSD attached with a SATA/USB3 adapter and that's been working ok, but with even SSDs the max power demand varies quite a lot, I'd guess HDDs can be even worse.

zagrim, 2021-05-21:
I'm also interested in the possibility of having a PIR sensor to trigger video capture for exactly the same reasons you stated, so at least I would very much welcome any follow-ups on that kind of projects also here.

As for the original issue you reported, the ordering issue might be possible to fix together with the "re-ordering of cameras" feature that has been requested a few times, but that depends how the config file updates are implemented (which I don't have the energy for the check). However, if the numbering of the camera configuration files happen to match (or at least their order matches) how the cameras are shown in ME GUI, your code could read `motion.conf` and perform a mapping between the IDs of the cameras in ME and their corresponding IDs in Motion webcontrol interface (like in your case camera 2 in ME would map to camera 1 in webcontrol URL etc).

MrDomingo, 2021-05-21:
I am not averse to making my solution public. My first prototype had a couple of errors but that was easily overcome and design has been updated. My design consists of one or more controller boards that would be strategically positioned around the house (possibly in roof space) and each controller connects via cables to one or two sensor boards.  Each sensor board has a Panasonic PIR and a Radar module.  The board has a DIP switch (12 pin/6 switches) and 3 of switches are to identify the station number.  That number dictates what cameras it is to support so it can trigger motion detection on these cameras only.  The ESP32 MCUs I purchased have an option to connect a proper antenna to it.  The controller would be housed in a metallic case and so external antenna is required.  Once I am happy with the sensors and how they work I will design a solution using ethernet and POE.  Btw I have also added a provision for the Dallas DS18B20 temp sensor, both for main controller board and also for sensor board.  That way the MCU may ignore PIR when temperature is too high.  In fact, the sensor board has provision to switch off power to either PIR or Radar module.  The main controller can also control 3 power lines (12V) so that one may want to switch on IR or normal LED illumination on the outside, or anything else.  Dahua camera is one of their 'starlight' models and bit of extra normal illumination may have the camera capture scenes in color during any night time incident rather than B/W IR images.  Solution is not cost optimized but first and foremost I did not want unreliable PIR or Radar triggers and I wanted main controller board to be useful for other projects.

MrDomingo, 2021-05-21:
As far as RPI issues are concerned, I did have external power for HDD Sata to USB3 adaptor but not for SSD's Sata to USB3 adaptor.  RPI4 has USB3 connector for power and I used recommended power adaptor for it.

MrDomingo, 2021-05-21:
I suspect RPI problem is related to particular SATA to USB3 interface and/or RPI system software not robust enough for these kind of deployments. What I also found was that using a command to get RPI to restart resulted in HDD not being recognized after restart.  I had to explicitly shutdown RPI4, and power off the HDD Sata to USB3 interface (and RPI, as there is no reset button) and then power up everything.

starbasessd, 2021-05-21:
The RPi4 issue of not recognizing USB drives has been updated in the latest eeprom updates. There is a setting about if and how long power is turned off to the USB ports during a reboot, where before it left the USB ports on, and improperly detected connected devices. I had similar issues with GPS on the USB ports until that update. They have found some USB SSDs took too long to 'spin up' and they put in a setting for a delay to let them be detected, too.
Neither here nor there, if you've switched to NUC, which I can't support as I don't have one. I can do any non-CM Pi, and any flavor Linux (on a real nice VBox virtual machine). Good luck with your use case. It sounds interesting.

MrDomingo, 2021-05-21:
@starbasessd, thank you for that update.  I have not given up on RPI4. Its fast enough for recording videos if it does not have to do motion detection itself.

starbasessd, 2021-05-21:
https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware/release-notes.md

JamBalaya56562, 2026-11-10:
Closing as a duplicate of #1887: both come from the order of the `camera` lines in `motion.conf`, which changes when a camera is disabled and enabled again, so Motion's own numbering (webcontrol, `monitor_N`) no longer follows motionEye's camera ids. motionEye's own buttons map the ids itself.

closed 2026-11-10 by JamBalaya56562