motioneye-project/motioneye #2125: manual install pillow issue....

hs6666, 2021-05-19:
Hi,

Following these instructions... https://github.com/ccrisan/motioneye/wiki/Install-On-Debian.... to install on a fresh install of debian buster (actually armbian running on a bananapi M1)

Eberything works well, but get pillow errors whilst running "pip install motioneye"

So tried the following fix before running pip ...... "apt-get install python-pillow", but it installs a different app....

*********************************************************************
root@bananapi:/home/user# apt-get install python-pillow
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'python-pil' instead of 'python-pillow'
The following additional packages will be installed:
  libimagequant0 liblcms2-2 libwebpdemux2
Suggested packages:
  liblcms2-utils python-pil-doc python-pil-dbg
Recommended packages:
  python-olefile
The following NEW packages will be installed:
  libimagequant0 liblcms2-2 libwebpdemux2 python-pil
0 upgraded, 4 newly installed, 0 to remove and 0 not upgraded.
Need to get 599 kB of archives.
After this operation, 1,696 kB of additional disk space will be used.
*******************************************************************************

I run pip again but I still get the pillow error, any ideas how to fix this issue please?

Thanks

PS motioneye still seems to run ok, what problems will i get thoguh becasue of the pillow error during pip.....

starbasessd, 2021-05-19:
Please run the commands:
python --version
pip --version
and report the results.

hs6666, 2021-05-19:
Hi, as requested :) .....

# python --version
Python 2.7.16
# pip --version
pip 18.1 from /usr/lib/python2.7/dist-packages/pip (python 2.7)

starbasessd, 2021-05-19:
There shouldn't be any issues if either python-pil or python-pillow installed. Both are the same, but there are upstream issues with the python 2.7 now that it's way beyond End Of Life. If you can access everything, you should be good to go. If you have any issues, be sure to ask here.

hs6666, 2021-05-19:
everything does seem to be working ok, thanks

JamBalaya56562, 2026-11-06:
Closing as solved, since you confirmed that everything works. That wiki page and Python 2 no longer apply: motionEye runs on Python 3 since 0.43.1 (#1978), and the current steps are in the [README](https://github.com/motioneye-project/motioneye/blob/main/README.md). With 0.44.0 on Debian 11 in an x86_64 container, Pillow installs without errors; ARM boards like the Banana Pi were not checked. Please reopen if it still fails there.

closed 2026-11-06 by JamBalaya56562