motioneye-project/motioneye #2127: Image corruption on network cameras

JYPDWhite, 2021-05-20:
Hi,

I'm dealing with corrupt images. 
[image]
I have a Camera from Reolink which is connected via wifi to my network. But sometimes the image is corrupt as you can see in the screenshot I provided.

I'm using the current build from dockerhub. I tried in on 2 machines one is running Debian Buster and the other is running Arch Linux (both recently updated). I also checked if it is a load issue so I started to generate a timelapse and watched if the preview got pixelated but no.) So I don't think it's a load issue. 

I also tried this additional motion parameters:
```
netcam_keepalive=force 
netcam_use_tcp=on 
 ```

This is the preview when it is working correctly:
[image]

Does anyone has an idea what I could try to solve this issue?

starbasessd, 2021-05-20:
Which reolink camera, and what URL are you using?

JYPDWhite, 2021-05-20:
It is a RLC-511W.

here is the config with all parameters (username and password are example values)
`` 
# @webcam_resolution 100
# @upload_subfolders on
# @upload_server 
# @enabled on
# @network_server 
# @motion_detection off
# @upload_port 
# @upload_location 
# @preserve_movies 0
# @network_username 
# @upload_movie on
# @id 3
# @webcam_server_resize off
# @upload_password 
# @manual_record off
# @upload_method post
# @upload_picture on
# @working_schedule_type outside
# @network_password 
# @upload_service ftp
# @preserve_pictures 0
# @storage_device custom-path
# @manual_snapshots on
# @network_share_name 
# @upload_enabled off
# @network_smb_ver 1.0
# @working_schedule 
# @clean_cloud_enabled off
# @upload_username 

threshold_maximum 0
netcam_userpass admin:
stream_quality 85
threshold 31984
noise_level 31
smart_mask_speed 0
pre_capture 30
movie_codec mp4
noise_tune on
stream_maxrate 20
netcam_url rtsp://reolink1:554/h264Preview_01_main
stream_localhost on
text_changes on
movie_filename %Y-%m-%d/%H-%M-%S
movie_max_time 0
lightswitch_percent 0
movie_passthrough off
event_gap 5
auto_brightness off
stream_port 8083
rotate 0
stream_auth_method 0
threshold_tune off
framerate 15
emulate_motion off
movie_output off
picture_quality 85
snapshot_filename %Y-%m-%d/%H-%M-%S
despeckle_filter 
snapshot_interval 60
minimum_motion_frames 20
stream_motion off
target_dir /var/lib/motioneye/Reolink1
movie_output_motion off
post_capture 30
stream_authentication viewerusername:viewerpassword
on_picture_save /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" picture_save %t %f
on_movie_end /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" movie_end %t %f
text_left 
picture_output_motion off
picture_filename %Y-%m-%d/%H-%M-%S
text_scale 1
locate_motion_style redbox
locate_motion_mode on
netcam_keepalive off
movie_quality 75
picture_output off
on_event_end /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" stop %t
text_right 
on_event_start /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" start %t
camera_name Reolink1
netcam_tolerant_check on
height 1920
mask_file /etc/motioneye/mask_3.pgm
width 2560

netcam_keepalive=force 
netcam_use_tcp=on 
```

starbasessd, 2021-05-20:
Did you try the 
rtsp://reolink1/h264Preview_01_sub
rtmp://reolink1:1935/bcs/channel0_main.bcs?token=[TOKEN]&channel=0&stream=0
Or try UDP instead of TCP?
Here's a link to using the rtmp setups, seems to be an issue with the high resolution reolinks across a number of applications:
https://forums.zoneminder.com/viewtopic.php?t=25874&start=15#p108502

JYPDWhite, 2021-05-20:
Good idea, I'll try it and see if this helps.

cap9qd, 2021-05-20:
I had the same issues with that model reolink camera and the rtmp link
cleared up my problems. I've been running 5 of them for around 6 months now
without any more issues.

terragit, 2021-05-31:
I had similar problems with a Realink RLC-410-5MP. RTMP works much better than RTSP, thanks.

JYPDWhite, 2021-06-17:
After several weeks of testing I can confirm RTMP is the solution. Thanks for the tip.

closed 2021-06-17 by JYPDWhite