motioneye-project/motioneye #2128: Can't upload any images when the IP automatically change after

eddy131415, 2021-05-23:
I can't received any pics when the IP automatically changed, than I tried to re-stall the .img and set the static IP or wifi private IP, but it didn't uploaded too, Accessing the upload service failed.
where's the problems?have anybody can help me?

starbasessd, 2021-05-23:
Are you using motionEye or motionEyeOS?
If motioneye, what is your base OS and what hardware/PC?
If motionEyeOS, which version, on what hardware?
How are you entering the IP address? 
Can you ping the IP address or view the WebGUI when set to the static IP?

Please answer ALL the questions in order to be able to help you better.

eddy131415, 2021-05-23:
Hello, thanks for your reply. I'm using motionEyeOS with raspberry pi 4.
The motionEyeOS version that I using is dev20201026.I tried to use the
cable network(have set up the static IP) and wireless but it didn't work
either. If any questions can't understand, please reply me.

Thanks a lot !

starbasessd, 2021-05-23:
Can you ping the IP address or view the WebGUI when set to the static IP?

starbasessd, 2021-05-23:
Sorry, clicked the wrong button. It's re-opened.

Can you ping the IP address or view the WebGUI when set to the static IP?

eddy131415, 2021-05-23:
Thanks for your reply. I try to release the 4 webcam and set the upload
step by step,they can work now!

Thanks a lot!!!!!

closed 2021-05-23 by starbasessd