motioneye-project/motioneye #2131: memory usage rises to 90% when recording

springnick1, 2021-05-31:
hey guys, i'm using motioneye in an Ubuntu VM on my proxmox host, with 4 ip cams. One of these cams is connected through rtsp, the other through rtmp since they are reolinks.

When motioneye starts recording one of the cameras, the memory usage goes up to 90% and stays at this point. Even when i manually stop recording, the memory usage doesn't go down and i need to reboot the machine. Without recording the memory usage is around 1.5 GB. A while back i had already running a motioneye machine that recorded 24/7 without any issues.

VM specs:  8 Cores, 6 GB ram

motionEye Version | 0.42.1
Motion Version | 4.2.2
OS Version | Ubuntu 20.10

starbasessd, 2021-05-31:
Motion/motionEye on i3 running on Ubuntu 20.04 (production)
motionEye Version | 0.42.1
Motion Version | 4.2.2
OS Version | Ubuntu 20.04
6 cameras, 3 d-link dcs-5020-L, 2 FosCam rtsp, 1 USB webcam
Welcome to Ubuntu 20.04.2 LTS (GNU/Linux 5.8.0-53-generic x86_64)
kevin@camera-server:~ $ free -h
              total        used        free      shared  buff/cache   available
Mem:          5.6Gi       785Mi       350Mi        13Mi       4.5Gi       4.6Gi
Swap:         2.0Gi       150Mi       1.9Gi
Tasks: 217 total,   4 running, 213 sleeping,   0 stopped,   0 zombie
%Cpu(s): 55.5 us, 19.9 sy,  0.0 ni, 22.4 id,  0.8 wa,  0.0 hi,  1.3 si,  0.0 st
MiB Mem :   5755.7 total,    129.8 free,    788.8 used,   4837.1 buff/cache
MiB Swap:   2048.0 total,   1896.8 free,    151.2 used.   4677.3 avail Mem

    PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND
1155422 root      20   0 2736256 280496  41796 S 234.6   4.8   5232:09 motion
    618 root      20   0  298020  44028  11652 S  11.0   0.7   2486:42 meyectl

I don't see that issue running in real world hardware.
Will build a VBox VM with similar specs and seewhat I find...

starbasessd, 2021-05-31:
Is your Ubuntu 20.04 Desktop or Server?

springnick1, 2021-05-31:
server. I previously had it running on debian also.

starbasessd, 2021-05-31:
I do not have any issues with a Debian 10.9.0 i386 VBox VM, either.
New Ubuntu 20.04 server being built right now, 8 cores, 6GB/50GB. Will update when completed.

springnick1, 2021-05-31:
i think i got the issue, which isnt really an issue. Proxmox shows the buffered/cached memory of the ubuntu machine as used if i understand it correctly. So its not really in use. Strange thing is, my prevoius motioneye VM and the other VMs are shown correctly.

starbasessd, 2021-05-31:
I have had nothing but issues with ProxMox and running motionEye in any VM.
I prefer VirtualBox, VMWare, KVM, M$ Hypervisor, or Docker...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

springnick1, 2021-05-31:
so far I was running fine with proxmox, but as I need to rebuild my system soon I  may switch to another Host-OS/Hypervisor.
But anyways, thank you, for your effort!

starbasessd, 2021-05-31:
Whatever system you choose, I'd go with a dedicated Lite or Server Host, for best results. My current VBox Host is a Ryzen 7/32GB/10TB running Ubuntu 20.04 LTS Server, and is currently supporting 6 full time VMs and whatever I build to support users an motion/motionEye...

springnick1, 2021-05-31:
Was thinking about switching motioneye to a dedicated server, guess I will give it a try.

starbasessd, 2021-05-31:
I picked up some refurbed Lenovo M73 TFFs, tossed in 8 and 16 GB / 1TB  and
left one a 6GB (2+4) and Ubuntu 20.04 and use one for my production
motionEye server feeding to a NAS. Really small, easy to hide in the house.
Cost me $50 @ 2 years ago. Well worth the money to look for the TFFs, Dell
and a couple other brands sell them, too.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

springnick1, 2021-05-31:
sounds very nice, which cpu are you using in these systems?

starbasessd, 2021-05-31:
i3s and i5s. Mixed lot...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-05-31:
I just applied a backup of my production box to my new VBox VM of Ubuntu 20.04 8 core / 6GB / 50GB and am getting similar numbers for memory usage / cpu usage...

springnick1, 2021-06-01:
sounds interesting,

another question which movie format do you suggest to use for recording?

starbasessd, 2021-06-01:
I avoid omx except On PIs. I mostly use h264 or mp4

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

JamBalaya56562, 2026-11-08:
Closing, as the reporter found the cause: Proxmox counts the buffered and cached memory of the Ubuntu VM as used, which Linux frees as soon as it is needed, so the memory was not really taken by motionEye or Motion.

closed 2026-11-08 by JamBalaya56562