motioneye-project/motioneye #2132: 'run a command' on motion detection does not update for new detection.

UranKhatolaaa, 2021-06-03:
Hi, I have 4 RTSP cameras connected to motionEye. I am using "run a command" on motion detection to start a process, which works fine. The problem is that if motion is detected in camera 1, motionEye executes, "run a command". A couple of seconds later when motion is also detected in camera 2 (I can see the red bounding boxes) but "run a command" for camera 2 is not executed. In my case camera 1 sees higher activity, and it is perpetually blocking the motion notification. If camera 2 (or 3 or 4) has an occasional activity, motion notification is never sent out. Motion notification from other cameras only works if camera 1 has no activity at all. 

Is there a way to change it so that the motion notification is always updated for the most recent motion detection and not wait until the detection from one of the cameras ends?

starbasessd, 2021-06-03:
Since Motion Detection is per camera, do you have the Run A Command  set up in both (all) Motion Notification setups?
Which hardware? Pi or other, Camera type, etc?
Base OS? 
Which install instructions followed?
Which version of motion?
Which version of motionEye?

UranKhatolaaa, 2021-06-03:
yes,  Run A Command is set up in Motion Notification for each camera. Motion-activated Still images, Movies all work fine. Motion notification works as well, but for only one camera at once. As long as there is motion in one camera, motion in other cameras is detected but Run A Command is not executed

It's running on Nvidia Jetson nano with 4 RTSP cameras (setup in motioneye using network camera option),
OS: Ubuntu 18.04
motion version: 4.0-1 installed using, sudo apt-get install motion
motioneye version: 0.42.1, Installed using  "pip install motioneye", with steps as described here,  https://github.com/ccrisan/motioneye/wiki/Install-On-Ubuntu-%2819.10-or-Earlier%29
inside, /etc/motioneye/ I can see camera-1.conf  camera-2.conf  camera-3.conf  camera-4.conf, they look like:
...
stream_auth_method 0
threshold_tune off
framerate 10
emulate_motion off
snapshot_filename
output_pictures off
despeckle_filter EedDl
snapshot_interval 0
minimum_motion_frames 2
stream_motion on
target_dir /home/safe/motion/media/usb-drive/Cam2
text_double on
post_capture 1
lightswitch 75
stream_authentication user:
output_debug_pictures off
on_picture_save /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" picture_save %t %f
on_movie_end /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" movie_end %t %f
text_left
picture_filename
locate_motion_style redbox
locate_motion_mode on
netcam_keepalive off
max_movie_time 10
on_event_end /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" stop %t; redis-cli SET cameraNum "0"
text_right neTra
on_event_start /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" start %t; redis-cli SET cameraNum "2"
camera_name Camera2
netcam_tolerant_check on
height 480
mask_file
width 640

The only difference in conf files for each camera are in on_event_start "redis-cli SET cameraNum "2"" and  camera_name  Camera2.

starbasessd, 2021-06-03:
You are using an ancient version of motion. I would upgrade to 4.3.2-1 first:
wget https://github.com/Motion-Project/motion/releases/download/release-4.3.2/bionic_motion_4.3.2-1_armhf.deb
If your Ubuntu 18.04 is the 64 bit version, change _armhf.deb to _arm64.deb
Install using:
sudo dpkg -i ./bionic_motion_4.3.2-1_armhf.deb
(again, if your Ubuntu 18.04 is the 64 bit version, change _armhf.deb to _arm64.deb)
reboot
Does your process lock any files? Create any files? I use Run A Command that plays an MP3 file as a 'doorbell' when people approach my front or back door. It sometimes concurrently.
If it doesn't (or even if it does) you might want to try the suggestions here:
https://www.tecmint.com/run-linux-command-process-in-background-detach-process/
If it does lock or create files, you may want to consider 4 copies of your command, modify as needed (changing file names, etc) and call each version separately.

UranKhatolaaa, 2021-06-18:
I updated motion package

The problem was kind of locking because I was using a single key with different values for different cameras. Motion sets a value for the key when the motion starts and as other cameras use the same key, nothing appears to happen if there are concurrent activities in other cameras until the motion ends and the key value is reset. 

So you gave me the right direction to proceed. I created separate keys for each camera and then just scanned them in a list in python. 

Many thanks.

starbasessd, 2021-06-18:
Sounds like it is working, great! If resolved, don't forget to close.

closed 2021-06-18 by UranKhatolaaa