motioneye-project/motioneye #2134: remote configuration changes

sebPomme, 2021-06-03:
Hi, 

Do you think it is possible to create some web endpoint to:
- `<GET=get | POST= enable| DELETE=disable>` notification `/api/v1/notification`
- `<GET=get | POST=enable | DELETE=disable>` motion detection `/api)/v1/motion_detection`
- Or some new idea ?

I think this can be useful to let another system (like homeassistant or jeedom or a custom script with curl) to enable / disable this features when no one is at home, or some more complex senarios.

starbasessd, 2021-06-03:
Look at the document for motion:
https://motion-project.github.io/motion_config.html
Look for section beginning "Web Control"
Some of those features already exist in the motion back end.

zagrim, 2021-06-04:
There have been others requesting this kind of thing, too, see #800 and #1223 and also #203 and #229. There are some ideas and workarounds in the comments.

sebPomme, 2021-07-08:
@zagrim @starbasessd  
Verry usefull links thanks. I think we can close this one for the moment.

closed 2021-07-08 by sebPomme