motioneye-project/motioneye #2136: Trouble adding new local cameras inside docker image

samohtred, 2021-06-09:
Hi Ccrisan,

Thank you very much for the whole lot of helpful comments and hints. However there's one thing which I haven't found on the internet so far. I am running MotionEyeOS inside docker under a 64 Bit Ubuntu version (should be latest version) on a laptop. However, when I start the local Webcams in Firefox, it works. When I try to open them in Motion or inside the MotionEyeOS environment, it fails. Do you have any hint for me ?

Best regards,
Thomas

starbasessd, 2021-06-09:
You have to add a volume pass through to your docker run command:
-v /dev/video0:/dev/video0
This should resolve it. There are issues with more than 1 USB camera attached. They need to be attached to different USB Root hubs.
You can search for 'docker usb webcam pass through' or 'docker pass through /dev/video0'

samohtred, 2021-06-09:
@starbasessd : Thank you very much for that hints. It helped pretty much. Now all cameras can be used inside the MotionEyeOS container. However, users should note that the "--privileged" option has to be set additionally.

starbasessd, 2021-06-09:
Don't forget to close if this resolved your issue. Glad to help...

closed 2021-06-09 by samohtred

lexhick, 2021-06-09:
Hi and sorry but I've the same problem to detect my local camera and I don't know how to add or modify my "Docker run command" in OMV5
my container is ok but I dont know how to modify it 
Thank you for help .

starbasessd, 2021-06-09:
I have no idea with Portaner, Several others have reported problems with
it, too.
If you follow the instructions to create the docker run command, without
using portainer, it works just fine.
You may want to see if there a portainer support group and ask there...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

samohtred, 2021-06-10:
@lexhick : If you want to modify a MotionEyeOS system, you'll have to unlock first since it is readonly. Look at this link : https://github.com/ccrisan/motioneyeos/wiki/FAQ#why-do-i-get-read-only-file-system-when-trying-to-edit-a-file