motioneye-project/motioneye #2137: email not sending because SSL/TLS key too short

aagaag, 2021-06-09:
The title says it all. My provider requires TLS and a strong key. I guess that the key automatically provided by motioneye is not sufficiently strong. How can I fix that?

starbasessd, 2021-06-09:
Or, it may be that the email server is using a too short of a key, due to the latest openssl update.
https://stackoverflow.com/questions/36417224/openssl-dh-key-too-small-error
This may help. I did many searches for sendmail.py fixes, but wasn't able to find anything. The CipherString solution might work, but I don't have a server to test against. GMail (one of the toughest out there), works fine for TLS mode.

JamBalaya56562, 2026-11-10:
Closing, as this got no follow-up. motionEye sends no key of its own; that error is OpenSSL refusing the mail server's too short key, which only the provider can change. Please open a new issue with the full error from `motioneye.log` if sending still fails with a current release.

closed 2026-11-10 by JamBalaya56562