motioneye-project/motioneye #2138: Unable to login to RTSP stream with Motioneye

rohtua, 2021-06-12:
Hi, I'm trying to get MotionEye (installed on ubuntu server) to connect to an RTSP camera I have. I can view the camera fine in VLC, Home Asistant and ONVIF Device Manager. However if I add the camera to MotionEye then I just get the grey camera symbol and no image. Looking in the MotionEye log it says that its failed to authorize the connection. I'm using the correct credentials as it works everywhere else (VLC, Home Assistant etc.) but not with MotionEye. 

I've tried setting the `rtsp_uses_tcp off` in the config, no luck. I've tried removing the authentication from the camera still no joy. I've tried adjusting the frame rates and resolution in MotionEye to match the camera in case that was causing an issue. I've used the rtsp URL that ONVIF device manager displays and the direct URL (i.e. just the IP address (which does work in VLC). Nothing seems to work I just get the blank box with a camera symbol in. On the camera itself the is a log which shows a login entry from the MotionEye server but it then logs out again immediately and this is recorded as a failed authorization by MotionEye. 

Any help would be much appreciated, my MotionEye log is attached. 

[motioneye log.txt](https://github.com/ccrisan/motioneye/files/6642434/motioneye.log.txt)

starbasessd, 2021-06-12:
How are you entering the URL, username, and password?
Which camera manufacturer and model are you using?

[image]

rohtua, 2021-06-12:
Hi thanks for replying, the camera is a generic unbranded one. Here's the link to it: 

https://www.rfconcepts.co.uk/smart-camera-3mp-3-6mm-3123.html

The model number on the box says SKU 3123 if thats of any help.

I've tried just about every combo I can think of with the IP username/password in the section you've screenshot.

Below are the combinations I've tried, I tested them all multiple times with authentication on and off on the camera. Just as a bit of a side note I did change the default RTSP port on the camera to 2565 from 554 but again I've tried to connect using both ports.

URL `rtsp://192.168.1.25:2565`

Where username and password fields are filled in; also tried it without filling in username and password

URL `rtsp://username:password@192.168.1.25:2565`

Tried this both with the username/password fields filled in and leaving them blank

URL `rtsp://192.168.1.25:2565/channel=0_stream=0.sdp?real_stream`  <--- this is the URL that ODM shows

I've tried this URL both with the username and password filled in in the fields and including them in the URL.

starbasessd, 2021-06-12:
If it were me, I'd reset the camera to :544
Then use 
URL rtsp://192.168.1.25:554/channel=0_stream=0.sdp?real_stream
Whatever username you have set into Username
Whatever password in Password
If you get the option for TCP or UDP, try UDP
Where did you get the info from ODM? What is ODM?
What URL did you use for VLC?

rohtua, 2021-06-13:
Hi,

I've used all the above URL's for VLC and they all work. If I don't include the username and password in the URL then VLC will prompt for them but each variation of the URL works. 

ODM is ONVIF Device Manager - its useful because it shows the full RTSP URL when viewing a feed rather than just the IP and port. 

I've already tried reseting to :554 and it still doesn't work. I do get the option for UDP as well but again no luck :( 

Everything I've tried just returns the failed to authorize message in the log and I can't work out why. The camera log just shows a login then immediate logout when I try to connect via MotionEye, and MotionEye just shows it as failed to authorize.

starbasessd, 2021-06-13:
For testing purposes, have you turned off authentication for the camera, or
tried a username of admin with a password of admin on the camera and then
add the camera with the basic settings?
Can you connect to it via an http: address for video or snapshots in
motionEye? jpeg and mjpeg or just as a network camera?

I am really at a loss.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

rohtua, 2021-06-13:
Yes I've tried those, and I've even tried creating new users on the camera both with full admin privilages and with view only. Unfortunately it's RTSP only. I can view the feed over HTTP but only in the cameras own web UI.

edashbolt, 2021-06-18:
I also have this issue. The weird thing for me was it used to work just fine, then one day I changed the setting for "Motion Optimization" to on and it just stopped working after that. I have deleted and re-added the camera, reinstalled MotionEye, nothing seems to work for me anymore. I can still access the RTSP stream in VLC but no luck with MotionEye, just "unable to open video device"

JamBalaya56562, 2026-11-08:
Closing, as this was with motionEye 0.42.1, and the later reports here were about different setups (Portainer, a 0.15.1 image). The failed authorization comes from Motion, which opens the RTSP URL with FFmpeg; when VLC and ONVIF Device Manager log in with the same URL and credentials, the camera may want a digest login that the Motion of that time did not handle. Please open a new issue with `motion.log` if the camera still cannot log in with a current motionEye and Motion.

closed 2026-11-08 by JamBalaya56562