motioneye-project/motioneye #2139: Constant saving of jpg images

opandroids, 2021-06-13:
Im running the lastest version of motioneye (0.42.1) my main issues even with motion detection disable and still images disabled 4 still jpegs are continously saved to my sftp server and the camera's sd card. Even with picture uploads disabled a constant stream of jpegs are uploaded.

Recording mode is on continuous recording with a max movie length of 1800 seconds.

starbasessd, 2021-06-13:
When you say "still images disabled 4 still jpegs are continously saved" what do you mean?
Are you saying 4 images are saved for every 30 minute video? 
Do you have Setting, Still Images set to Off?
If you have more than 1 camera, do you have them all set that way?
Under File Storage, Upload Media Files, do you have Upload Pictures turned on?

opandroids, 2021-06-13:
> When you say "still images disabled 4 still jpegs are continously saved" what do you mean?
4 images ever 60 seconds are saved. on both sdcard and sftp server even though image upload to sftp is disabled. Motion detection and still images are disabled.
> Are you saying 4 images are saved for every 30 minute video?
No video's are uploaded to sftp simply 4 still images per minute.
> Do you have Setting, Still Images set to Off?
Yes still images are set to off
> If you have more than 1 camera, do you have them all set that way?
1 Camera only
> Under File Storage, Upload Media Files, do you have Upload Pictures turned on?
File storage has both SD card and sftp set. Image uploads are disabled. But pictures are still uploaded and stored.

Here are a few pictures. https://imgur.com/a/kcB5KvC

starbasessd, 2021-06-13:
Interesting.
What system (base OS) are you using?
Which install instructions did you use? 
Are you using Docker, or HomeAssistant or other wrapper?

opandroids, 2021-06-13:
> What system (base OS) are you using?

Latest Raspbian 
> Which install instructions did you use?

Latest Raspbian
> Are you using Docker, or HomeAssistant or other wrapper?

Yes im its running in docker

starbasessd, 2021-06-13:
Which Docker? apt install docker.io, or apt install docker-ce?

starbasessd, 2021-06-15:
Which Docker? How did you install Docker? apt install docker.io, or apt install docker-ce?

starbasessd, 2021-06-15:
What is your docker run command?

yurnov, 2021-09-25:
Hi @opandroids, you have mentioned that some jpeg stored in SD card inserted in your camera. Usually, storing of still images/movies on internal SD can be configured directly in the camera, also, a lot of IP camera models are able to upload files to external services, like FTP, SFTP or SMB servers.

So, could you please check, maybe something like periodical still images storing/uploading is occasionally configured directly in your camera config?

starbasessd, 2021-09-25:
@yurnov I hope you get a reply, there was nothing for the past 3 months...

yurnov, 2021-09-26:
Hi @starbasessd, I think you mentioned me instead of @opandroids who has an issue. I  just tried to propose some troubleshooting.

Anyway, yep, it's 3 months without reply, so, probably this issue can be closed

closed 2021-09-26 by starbasessd