motioneye-project/motioneye #2140: Make MotionEye docker multiplatform by default

Slyke, 2021-06-15:
Would it be possible to compile the MotionEye docker image to work on the common ISAs without having to specify the instruction set as a tag?

For example, when pulling this for a Raspberry Pi, you must specify `master-armhf` as a tag or it'll pull the x86_64 docker image by default.

It's possible to do with the buildx command: https://docs.docker.com/desktop/multi-arch/ but I'm not sure how difficult the implementation would be.

zagrim, 2021-06-15:
I wonder if this, too, is related to #1897 and the fact that all images are marked as "linux/amd64" regardless of the real target architecture.

Slyke, 2021-06-15:
@zagrim certainly possible!

JamBalaya56562, 2026-11-10:
Fixed by #2361, which builds the Docker image for several platforms in one manifest; `docker pull motioneyeproject/motioneye` now picks the right one for amd64, arm64, armv7 and more, without an architecture tag.

closed 2026-11-10 by JamBalaya56562