motioneye-project/motioneye #2144: Motioneye video

monit-org, 2021-06-23:
Hi, I want show videos,  not created by motioneye, on motioneye video output in browser. Is there any way to do that?

starbasessd, 2021-06-23:
You can upload to the FTP or SMB folders if using motionEyeOS, if they are the same format (usually mp4) they should show up when you drill through the videos.
You need to enable Write feature in Settings, Services for whichever service you want to use.
If you want to upload to another OS running motion/motionEye, that is beyond the scope of my support, you should be able to find instructions for adding the share and settings.

monit-org, 2021-06-23:
> You can upload to the FTP or SMB folders if using motionEyeOS, if they are the same format (usually mp4) they should show up when you drill through the videos.
I'm not using motionEyeOS, it only motionEye
> You need to enable Write feature in Settings, Services for whichever service you want to use.
Please can specify that in which serivce I have to do this,
> If you want to upload to another OS running motion/motionEye, that is beyond the scope of my support, you should be able to find instructions for adding the share and settings.

To be more specific, I'm making 1min video in motionEye, Sometime I want to view 30mint video, so I merge these 30 one mint videos into one, but the problem is that I can't view this 30mint merge video on motionEye video view section. I don't want to use FTP and other service to do that. Is it possible?

starbasessd, 2021-06-23:
Where are you editing/creating them? They would have to be in the camera path.

monit-org, 2021-06-23:
> Where are you editing/creating them? They would have to be in the camera path.

Yes, the video is in camera path.

starbasessd, 2021-06-23:
Do you not see the file? What format is the original, and what format is the 30 minute file?

monit-org, 2021-06-23:
> > Where are you editing/creating them? They would have to be in the camera path.
> 
> Yes, the video is in camera path.
> Thanks it workes.

closed 2021-06-23 by monit-org