motioneye-project/motioneye #2145: No recording when moving

RPiDespair, 2021-06-25:
Hi,

Configuration
Two RPi 3b
No. 1 RPi 3b with NoiR V2 camera
No. 2 RPi 3b with V2 camera

motionEye version 0.42
Motion version 4.3.1
OS version Raspbian 10 
Running As Normal User
Only after the problem occurred did I sudo apt-get upgrade 06/24/2021.

MotionEye no longer takes photos when changes are made in front of the camera. Take a snapshot works.
I am running two identical configured PIs with a camera and MotionEye. Only the cameras differ, one of them is for infrared recordings.
The MotionEye settings are identical for both PIs. MotionEye in PI with the infrared camera makes motion triggered recordings,
but MotionEye in the PI with the daylight camera do not take motion triggered recordings.
What could be the cause?

starbasessd, 2021-06-25:
Hard to say.
I would copy the /etc/motioneye/camera-1.conf file from the one that is working to the one that isn't working, and reboot it.
Another option: since one works, I would copy that SDCard to the one that doesn't work, and try that.
Do you see the video from the camera on the Pi that isn't working?
If you do, turn on Settings, Motion Detection, Show Frame Changes, and look for the red box around motion.
If you don't, in SSH or console, run the command
vcgencmd get_camera
and post the result.

RPiDespair, 2021-06-25:
Thank you for your suggested solution.
The exchange of the camera-1.conf was unfortunately not successful.
I cannot currently exchange the SD cards (nesting box).
I see the current live recording but not the red frame when moving.
vcgencmd get_camera
supported = 1 detected = 1

starbasessd, 2021-06-25:
If you are not using static IP addresses on the Pis,
On the Good camera Pi, in the WebGUI, go to Settings, General Settings, and do a backup.
On the Bad Camera Pi, in the WebGUI, go to Settings, General Settings, and do a Restore.
Do a reboot on the Bad Camera Pi. 
Check for motion.
If that doesn't work, it may be that something on the Bad Camera Pi is corrupted, and the fastest solution would be to re-image and re-build...

RPiDespair, 2021-06-25:
I am using static ip addresses.
The quickest solution would be to remap and recreate ...
Just reinstall motioneye or completely including the operating system?
What happened before! Please copy the following link into a new browser window .
https://translate.google.com/translate?sl=auto&tl=en&u=https://forum-raspberrypi.de/forum/thread/52245-raspberry-extrem-langsam-motioneye-funktioniert-nicht-mehr/?action%3DlastPost

starbasessd, 2021-06-25:
What is " Kamea "? 
I would start with a fresh OS, update, then install motion/motionEye using these instructions:
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian

RPiDespair, 2021-06-25:
Kamea = misspelling
Kamera = camera

I uninstalled motioneye and reinstalled it. Motion is recognized but is very unstable. When the birds have flown out I will copy my complete buster image from 2020 that already contains motioneye 4.3.1 onto the SD.
Thank you very much for your support.

JamBalaya56562, 2026-11-08:
Closing, as this came after an `apt-get upgrade` on Raspbian 10, the same camera config did not help, and the reporter planned to restore the earlier SD card image. The missing red frame shows Motion stopped seeing changes on that Pi, which points to the upgraded system there rather than to the settings. Please open a new issue with `motion.log` if a camera still records no motion with a current motionEye.

closed 2026-11-08 by JamBalaya56562