motioneye-project/motioneye #2146: MotionEye is flooding my /var/log/syslog file

miguelgilmartinez, 2021-06-27:
MotionEye writes hundreds of lines per minute like this next **when I open its web interface**:
GET /picture/2/current/?_=1624813388447&_username=myusername&_signature=1bb8189s72e3d9e3de46de1367 HTTP/1.1 (Mozilla/5.0 (X11; Linux x86_64) 

I have configured log_level as "quiet" and "error", restarted MotionEye but it keeps on flooding my log.
Current version: 9.1.4 on Home Assistant core-2021.6.6 and Docker 20.10.7

Thanks in advance.

starbasessd, 2021-06-27:
What is the URL you are using when it triggers the flood?

miguelgilmartinez, 2021-06-28:
> What is the URL you are using when it triggers the flood?

I'm using motionEye inside HomeAssistant, so URL is hidden. Just open motionEye addon and the flood is there. I close camera viewer, and it stops. If you take a look to that URL, the number after "/picture" is the camera id.

starbasessd, 2021-06-28:
Can you access motioneye directly? 
Use the URL http://ip_address:8765?
Or access the camera stream via the URL http://ip_address:8081?
If you can, does the log flood?
I know the docker app works fine, solo, the direct install works fine, solo, but we have MANY issues with HA listed here, and no help from them.

miguelgilmartinez, 2021-06-28:
> Can you access motioneye directly?
> Use the URL http://ip_address:8765?
Nope. It's running inside HA, I think.
> Or access the camera stream via the URL http://ip_address:8081?
I had that service down completely. I switch it on and issue persists. Sorry.
> If you can, does the log flood?
> I know the docker app works fine, solo, the direct install works fine, solo, but we have MANY issues with HA listed here, and no help from them.
If nobody here has this issue (solo version) maybe it's a HA issue. It's weird, because it happens clearly only when I open ME web interface, and it looks like a kind of NGINX's denial of service.  Maybe HA is filtering that petition, maybe HA is the process which floods the log... I set log_level: fatal and restart ME and still flooding.

starbasessd, 2021-06-28:
I will attempt a HA setup one more time, and report.
In the mean time, if you look at the docker run command here:
https://github.com/ccrisan/motioneye/wiki/Install-In-Docker
do you have the port 8765 opened? 
Do you have that port opened in nginx?

starbasessd, 2021-06-28:
Are you running HAOS, HAdocker, or HA on another OS?

miguelgilmartinez, 2021-06-28:
HAdocker on Raspbian.

starbasessd, 2021-06-28:
OK, are you using anything like Portainer?

starbasessd, 2021-06-28:
Stage 1 test:
Install & setup HAOS on RPi3B+, 32GB 
Add motionEye to HAOS from Community
Restore from current backup of motionEye Settings backup
3 cameras work (mjpeg Network cameras)
3 cameras fail to work (2 rtsp and 1 motionEye stream)
Delete and re-add motionEye stream, works
Delete and re-add Network Camera, URL rtsp://ip_address:88/videoMain as TCP, fails
Delete and re-add Network Camera, URL rtsp://ip_address:88/videoMain as UDP, fails
Nothing in logs.

starbasessd, 2021-06-28:
Which docker install? docker-ce or docker.io?
Can you check URL http://ip_address:28765 please? In most of the troubleshooting hints, it says to try that. (you could also try https://...)

miguelgilmartinez, 2021-06-28:
> Which docker install? docker-ce or docker.io?

Docker-ce 

> Can you check URL http://ip_address:28765 please? In most of the troubleshooting hints, it says to try that. (you could also try https://...)

Nope... this port is not open.

starbasessd, 2021-06-28:
Jumping back in... If you don't hear from me, send in the dogs...

miguelgilmartinez, 2021-06-28:
The point is this is a normal and legal operation: My browser is displaying live webcam images from motionEye so all these petitions are being logged in the syslog by (maybe) webserver. No error, no warning, just plain GET petitions correctly served to my browser and unnecessary logged in.

miguelgilmartinez, 2021-06-28:
> OK, are you using anything like Portainer?

I have Portainer installed too.

starbasessd, 2021-06-28:
OK, they all may be conflicting. I know the motionEye docker has issues with Portainer, I have issues in with them, especially with more than 1 camera. Checking the rest.

starbasessd, 2021-06-28:
RPi3B+, Install, update & Setup latest RPiOS: (5/7/2021)
Install docker-ce
Test 1:
Install motionEye docker
test motionEye (install backup configuration, make sure all cameras work, no log issues)
Uninstall motionEye docker
Test 2a
Install HomeAssistant docker
test HomeAssistant
Works, gives default cards and finds a TV
Test 2b
install motionEye Integration
Can connect to my primary motionEye Hub
Sees all 6 cameras
uninstall motionEye Integration
Test 3
install motionEye Docker
set up to duplicate my primary motionEye Hub
working
install motionEye Integration
connect to localhost docker motionEye
Sees 5 cameras, unable to see simple mjpeg camera streamed from primary motionEye Hub
No issues in logs... (having set yaml to log everything...)
I don't know if issue lies in nginx or _maybe_ portainer (I didn't set up either) but I confirmed system is working as above...

starbasessd, 2021-06-28:
What port are you using with nginx? https:// nginx is not installed by default with RPiOS/Docker/HADocker/mEDocker...
My Pi is on ip_address. HA is on ip_address:8123. mE is on ip_address:8765.

miguelgilmartinez, 2021-06-28:
> What port are you using with nginx? https:// nginx is not installed by default with RPiOS/Docker/HADocker/mEDocker...
> My Pi is on ip_address. HA is on ip_address:8123. mE is on ip_address:8765.

It's an out-of-the-box Home Assistant and MotionEye instalation. http://my_local_ip:8123 then Home assistant talks to MotionEye.

starbasessd, 2021-06-28:
How/where is you add motionEye? As an Integration? Or from a link?
Something isn't 'right'. My HA doesn't have any links to motionEye directly. I can see the cameras, if I click on one, I can get some settings, none of which go to motionEye. They all go back to the Integration page, or indicate it is an Entity or Device, but they don't have any available settings, either. To get to motionEye, I need to open a new browser, and the URL http://ip_address:8765 or if using the local docker version, I can use http://localhost:8765
What am I doing different? (other than I WON'T install nginx or Portainer, both of which can re-direct stuff unintentionally)

starbasessd, 2021-06-28:
I even added both hubs, the primary, and the local docker. Both work fine, no log issues. 
I re-read your original post, you're using core-2021.6.6, not the Docker version, is that correct?
Building a new SDCard...

starbasessd, 2021-06-28:
Got HA core installed on the Pi3B+ in the python virtual environment. It found a bunch of my other hardware that the other versions didn't, but didn't have motionEye in the built in list of Integrations. Went to the webpage to do the My install, and it got pissy about being python 2.7 and 3.7, and not 3.8. Now installing/compiling 3.8.

miguelgilmartinez, 2021-06-29:
My MotionEye is a Supervisor addon, like Nodered, DuckDNS and several others installed and not interfering with the logs.

starbasessd, 2021-06-29:
I'm sorry, but I am not able to reproduce your setup, and everything I've tried seems to work for me. Hopefully someone else will respond, with other ideas, but as of Right Now, with all the other stuff you have installed, I'm beginning to believe it's a conflict somewhere between the apps you are using.

miguelgilmartinez, 2021-06-29:
> I'm sorry, but I am not able to reproduce your setup, and everything I've tried seems to work for me. Hopefully someone else will respond, with other ideas, but as of Right Now, with all the other [...]

Probably you are right. I will keep an eye on that and let you know. Thanks for your help.

JamBalaya56562, 2026-11-08:
Closing, as these lines do not come from motionEye: it never logs successful `/picture/<id>/current/` requests, whatever the log level, and its request lines look like `200 GET /... 1.23ms`, without the browser's user agent. So they were written by something in front of it in the Home Assistant add-on, such as its web proxy, which belongs to the add-on's repository.

closed 2026-11-08 by JamBalaya56562