motioneye-project/motioneye #2147: mjpg client fails to make connection (repeats connecting and disconnecting)

takashikumagai, 2021-07-01:
I'm trying to run motionEye with Python 3 on a pi. I have been able to start the app, login, and access the UI from a browser (Chromium) on another computer on the same local network, i.e. I'm accessing the motionEye UI via an IP address like 192.168.1.23:8765. There seem to be a few things wrong with it and the app is not working well. One problem is that the mjpeg client is disconnected right after connected (within a second), then the connection & disconnection process is repeated ad infinitum. This happens the same way with a USB webcam and a raspberry pi camera, both of which can take snapshots with raspistill and fswebcam, respectively, so I am certain that cameras are detected and functioning.
```
   DEBUG: creating mjpg client for camera 1
   DEBUG: mjpg client for camera 1 connected on port 8081
   DEBUG: connection closed for mjpg client for camera 1 on port 8081
   DEBUG: mjpg client for camera 1 on port 8081 removed
   DEBUG: creating mjpg client for camera 1
   DEBUG: mjpg client for camera 1 connected on port 8081
   DEBUG: connection closed for mjpg client for camera 1 on port 8081
   DEBUG: mjpg client for camera 1 on port 8081 removed
   DEBUG: creating mjpg client for camera 1
   DEBUG: mjpg client for camera 1 connected on port 8081
   DEBUG: connection closed for mjpg client for camera 1 on port 8081
   DEBUG: mjpg client for camera 1 on port 8081 removed
   DEBUG: creating mjpg client for camera 1
   DEBUG: mjpg client for camera 1 connected on port 8081
   DEBUG: connection closed for mjpg client for camera 1 on port 8081
   DEBUG: mjpg client for camera 1 on port 8081 removed
```

Model: Raspberry Pi 3
OS: Raspberry Pi OS Lite (/etc/os-release is "Raspbian GNU/Linux 10 (buster)")
Kernel version: 5.10
Motion Version 4.3.1+git20210125-34091a7
motionEye Version	0.42
motionEye branch: python3
Python version: 3.7.3

This is what the UI is currently looking like:
[image]

starbasessd, 2021-07-01:
motionEye / motionEyeOS are dependent on puthon2.7 and pip2 to function correctly. Conversion to Python3 has not been completed at this time.

takashikumagai, 2021-07-02:
I see. Thanks for clarifying the situation. I might do some more digging myself to assess the practicality of running motionEye with Python 3.

In the meantime, I somewhat doubt if the error above is a Python 2 vs 3 issue; for one thing, I'm running motionEye as a user other than root (the user is in the sudo group but not the root user). Could it be the cause of the problem? It looks like this:

```
$ python3 meyectl.py startserver -c /path/to/config/file
```

starbasessd, 2021-07-02:
Did you follow these instructions?
https://github.com/ccrisan/motioneye/wiki/Running-As-Normal-User
When you installed motionEye, how did you get around the pip2 (lack thereof) issues?

starbasessd, 2021-07-02:
Created a Pi3B with RaspberryPiOS (05/07/2021) Lite, fully updated.
Installed motionEye using instructions here:
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
Set up Run As Normal User using instructions here:
https://github.com/ccrisan/motioneye/wiki/Running-As-Normal-User
Installed 5 of my networking cameras.
Working as expected.
Tell me where you found instructions for using Python3, please, so I may test it.

takashikumagai, 2021-07-03:
> Did you follow these instructions?
> https://github.com/ccrisan/motioneye/wiki/Running-As-Normal-User

Oh I didn't know about this guide, thanks. I'll definitely check this one out. Let me get back to you after that

takashikumagai, 2021-07-03:
I followed the instructions on the "Running As Normal User" wiki page and the connection error seems to have gone away. Thank you so much. I thought I was doing basically the same thing but after placing everything under my home directory (as shown in the instructions), things started working. So maybe something was wrong with my directory permission settings (run, log, or media);

> When you installed motionEye, how did you get around the pip2 (lack thereof) issues?

Right, since it was not possible to install motionEye for Python 3 using pip3, I simply cloned the repo and executed meyectl.py from a terminal. The gist of it is this:

```
$ git clone https://github.com/ccrisan/motioneye.git
$ cd motioneye
$ git checkout python3
$ cd motioneye
$ python3 meyectl.py startserver -c /path/to/config/file
```

There were some minor changes that I had to make in order to start the motionEye server with Python 3 but it was not a big deal. I might share those changes later.

As for dependencies, e.g. tornado, I installed them separately using pip3.

I haven't checked whether all the features of motionEye are working yet, but at least the USB webcam is now showing the live preview on the UI. Thanks a lot. On the other hand, my pi camera is not showing the preview and doesn't seem to be working, but I should probably create a separate issue for this.

starbasessd, 2021-07-03:
Good to hear you fixed the permissions issue.
Run command:
vcgencmd get_camera
and post the result in a new issue, please...

JamBalaya56562, 2026-11-08:
Closing, as the reporter solved it by following "Running As Normal User": the connection loop came from permissions on the run, log or media folders of the user running motionEye. motionEye runs on Python 3 since 0.43.1, so that part no longer applies either.

closed 2026-11-08 by JamBalaya56562