motioneye-project/motioneye #2148: picture resolution

aurel5959, 2021-07-02:
Hello everybody
Is it possible to take photo in 3840x2160 with official camera HQ
I use Raspberry PI4B  with raspberry PI os.
Thanks for your help

starbasessd, 2021-07-02:
Which version of motion?
Which version of motionEye?
Which OS?
(all found in Settings, General in WebGUI)
Processing motion detection is difficult at higher resolutions.

starbasessd, 2021-07-02:
You may want to look at this issue:
https://github.com/ccrisan/motioneye/issues/1764

aurel5959, 2021-07-02:
motionEye 0.42.1 (7 june 2020)
raspberriPIos 5.10.17

I've already read issue #1764 , but since it is over a year old I wanted to know if there are any changes.

starbasessd, 2021-07-02:
Which version of motion?

aurel5959, 2021-07-02:
sorry 
motion version 4.3.2 (26 oct 2020)

starbasessd, 2021-07-02:
Is the limit due to the dropdown being 1920x1080 or something else?

aurel5959, 2021-07-02:
i think it's something else
I select custom resolution and test 2560x1440 and 3840x2160 but I no longer have an output image

starbasessd, 2021-07-02:
Unfortunately, I don't have one to test. Can you get video at 1920x1080? Or lower?

aurel5959, 2021-07-02:
I can't test immediately, yes i can record video and take picture at 1920x1080 and lower  without problem

starbasessd, 2021-07-02:
I will leave this here in hope that someone else can post suggestions.

aurel5959, 2021-07-02:
thanks for your help

JamBalaya56562, 2026-11-10:
Closing, as this was with motionEye 0.42.1 and Motion 4.3.2 on the legacy Pi camera stack, and no one could test the HQ camera here. Please open a new issue with `motion.log` if a resolution above 1920x1080 still gives no picture with a current release.

closed 2026-11-10 by JamBalaya56562