motioneye-project/motioneye #2149: Spinning lens icons

mattpstanding, 2021-07-03:
Hi All,

I have been using ME and MEOS for years without problem. Recently my master ME installation has stopped working.
I have two MEOS cameras linked to this master ME, all work if I browse to their IP addresses, but if I try and browse to the master ME I just get constant spinning camera lense icons.
This on on a PI 3, fully patches OS and ME. Any pointers would be great!

Thanks in advance, Matt

starbasessd, 2021-07-03:
The master ME hub:
Can you get to the WebGUI, Settings, General? If yes: 
what version of motion?
what version of motionEye?
What OS?
Can you get to the console or SSH? If yes, please run the commands:
vcgencmd measure_temp
vcgencmd get_throttled
and post the results. Do this BEFORE any reboots.
If you can get to the console or SSH, but not to the WebGUI, also try this command:
sudo systemctl status motioneye
If you can access the logs, please attach the /var/log/motioneye.log

JamBalaya56562, 2026-11-10:
Closing, as this got no answer to the questions above and was with a 2021 setup. Please open a new issue with `motioneye.log` from the hub if its cameras still keep loading with a current release.

closed 2026-11-10 by JamBalaya56562