motioneye-project/motioneye #2150: Problem with pi camera (Python 3 branch)

takashikumagai, 2021-07-04:
Hi, starbasessd,

About the pi camera issue I brought up on #2147, here is what vcgencmd shows:

```
$ vcgencmd get_camera
supported=1 detected=1
```

But I realized that I had selected the wrong camera type. This "Local MMAL Camera" is the pi camera, right?

[image]

Once I selected it, the preview window started showing the live feed.

However, I did have to make the following change. Otherwise, I get a type mismatch error (subprocess.check_output returns bytes in Python 3).

https://github.com/takashikumagai/motioneye/commit/e416c1222b953f82900bc26faad5c0cf3218ea75

Thank you for helping me out again

starbasessd, 2021-07-04:
Yes, CSI (ribbon cable) connected cameras should be MMAL type cameras.
My request for the get_cameras result was to see if there was an issue with the camera connection. You found you had picked the wrong camera type, when corrected fixed the visibility.
Thanks for the python fix. Hopefully will be added to the rest of the 'fixes'...

zagrim, 2021-07-04:
@takashikumagai please create a pull request for that commit against the python3 branch in ccrisan/motioneye so that it could be possibly merged in.

takashikumagai, 2021-07-05:
@zagrim thanks for your suggestion. I've never created a pull request before so I just followed the instructions that appeared on the screen and created one anyway. Not sure if I did it the right way.

zagrim, 2021-07-06:
Seems right to me :+1:

MichaIng, 2022-03-19:
Fix in `dev`.

closed 2022-03-19 by MichaIng