motioneye-project/motioneye #2151: Motion Eye Screen white bar or full White Screen

gizzburn, 2021-07-04:
I keep getting this white screen on my pi, I have tried another pi, another screen and re-installed motioneye and raspberry image, but I still get the error. It can happen within 5 seconds or 5 mins. A quick reload of the screen and it works again but will randomly pop up this white bar or the full screen goes white.

This is the stream ip on separate pi using chrome, it is all thats on this pi. It does not show up on the same stream on desktop pc when the same stream is loaded.
[image]

starbasessd, 2021-07-04:
motionEye or motionEyeOS?
Version of motion?
Version of motionEye?
Version of OS?
(all 3 above can be found in WebGUI, Settings, General)
Version of Chrome?
(3 dots in upper right, Help, About)
What OS are you using on the Desktop PC?
What Browser on the Desktop PC?

JamBalaya56562, 2026-11-08:
Closing, as this got no answer to the versions asked for. The white bar showed only in Chrome on that Pi, while the same stream was fine on the desktop PC, which points to that browser or its graphics on the Pi rather than to the stream motionEye sends. Please open a new issue with the versions if it still happens with a current release.

closed 2026-11-08 by JamBalaya56562