motioneye-project/motioneye #2153: Set On-Screen Font Type and Color

smyers75, 2021-07-06:
The On-Screen Text is close to impossible to read with certain backgrounds (white color).
How can we change the Font Type and Color?

We know how to change the Font Size using Text Scale, but with the current Font, when it gets larger it gets so pixelated it is hard to read.  

Example "88" in a larger Text Scale looks like it could be 99, 98 89, 80, 90...

starbasessd, 2021-07-06:
It is not currently an option in motionEye or motionEyeOS.
IIRC, it's a motion feature (someone correct me please) and built into that program.
Here is a link into their documentation:
https://motion-project.github.io/motion_config.html#text_left
but there isn't any indication of being able to change the font or color.
You may want to ask in their support forums:
https://github.com/Motion-Project/motion/wiki/FAQ

JamBalaya56562, 2026-11-08:
Closing, as the on-screen text is drawn by Motion, and motionEye can only pass what Motion offers: the text itself and Text Scale. Motion has no option for the font or colour, so this request belongs to the Motion project (Motion-Project/motion).