motioneye-project/motioneye #2154: Chrome issue - no way to disable "lazy images" flag

misanil, 2021-07-08:
It seems that last updates of Crome do not allow to modify Its "flags".
So, is no more possible to disable the "lazy images" item, and Chrome (and other browsers too), again, does not show camera images! (Grey area with "rotating symbol").
Using android version, I have erased chrome updates in order to use again the "chrome://flags/#enable-lazy-image-loading" tool, but Chrome soon updates (and Is fine, in terms of security)....
Please, has anyone found how to fix this problem with new version of Chrome,  or has an idea about how to access to "chrome://flags/#enable-lazy-image-loading" disabling tool with new Chrome release?
Thanks
misanil

starbasessd, 2021-07-08:
Can you be more specific as to the version of Chrome, and what platform?

starbasessd, 2021-07-08:
I am running
chrome 91.0.4472.101
on 
Android 10; V350U build / QkQ1.200810.001
on an AT&T Maestro Plus

misanil, 2021-07-08:
on my PC and my Surface, I'm running chrome  91.0.4472.124 (Build ufficiale) (64 bit) connecting via VPN to Odroid with MotionEye

On my smartphone (Motorola One Vision - android 10 motorola one vision build QSAS30.62-33-15) i'M RUNNING  chrome 77.0.3865.92 

Thank you

starbasessd, 2021-07-08:
If the smartphone is having the issue, I'd see about upgrading it to something a bit more current. You shouldn't be having the issues with the PC or Surface.

starbasessd, 2021-07-08:
Chrome 77 came out in Aug 2019 (2 years ago) and current is 91, soon to be 92.

misanil, 2021-07-08:
sorry, you are right... on my smartphone I've this afternoon discarged chrome' updates in order to be able to use flags for lazy images; here attached the screenshot of flags page with this old version of crome on my motorola.

[image]

Here is the new page of flags with chome 91, from my PC

[image]

No more available the "lazy image" option 
Hope these two screenshots (old and new chome) will help
thanks

misanil, 2021-07-08:
and here attached the screenshot after chrome update (now 91.0.4472.120) on my motorola, made right now
The page is differente as the list of flags (missing "lazy images" option, as on my PC 
[image]
with chrome91)

starbasessd, 2021-07-08:
With the newer versions, I don't need the option for either motion/motionEye (on another OS) or motionEyeOS. That was an old issue long since resolved. 
What version of motion, motionEye, and odroid?

misanil, 2021-07-08:
I'm using Odroid C2 and this is the MotionEye version:

[image]

starbasessd, 2021-07-08:
Current version is 20200606 and current dev version is dev20201026. Please consider upgrading.

starbasessd, 2021-07-08:
You can find many issues resolutions in the motionEyeOS github and the current releases at
https://github.com/ccrisan/motioneyeos/wiki
and
https://github.com/ccrisan/motioneyeos/releases

misanil, 2021-07-08:
I'm doing updating via VPN... hoping do not crash...

misanil, 2021-07-08:
bad news... after 5 minutes no way to reconnect to odroid... next sunday I'll be back home and I'll shut down odroid (cutting power) trying to restart it
sigh

starbasessd, 2021-07-08:
Update function (Settings, General) only works for Raspberry Pis.

zagrim, 2021-07-09:
Upgrading should anyway fix the issue, since the fix for this issue is present already in MotionEye 0.41 (see e.g. #1694) and consequently in MotionEyeOS 20190911.

misanil, 2021-07-17:
Hy everyone, new issue... Flashing image version 20200606 for ODROID C2, that seems to be the last version available on web site, the system starts but the MotionEye application not. Probably this Is the reason why After automatic upgrading from software It stopped ti work.
Now I've flashed an older version (20190427) and It starts, but not showing again images on Chrome etc.
Mhhhh bad news

starbasessd, 2021-07-17:
If you do a fresh, clean install of 20200606 and test, be sure to clear cache & cookies, too in the browser. Sometimes when a motionEye / motionEyeOS uses the same IP address, it remembers some of the old settings for it until you clear cache and cookies... 
What error do you get? can you get to the logs? motion, motioneye, and messages?

misanil, 2021-08-05:
Another update after several trials...
My odroidC2 does not work with any release after the 20190911; after that release, the odroid seems to start (blu led starts to pulse) but doesn't really work.
I tried also to delete chrome's history, but it didn't work.
With release 20190911 I can finally see images with Chrome!!    But:
- the upload feature of previously saved settings doesn't work
- Reboot function causes a crash and forces me to flash again the software (no way to reboot the system in other ways)
- In case of power off from the grid, same crash of the software as above
Whereas ''Shut Down'' from the software is OK (the system can restart).
Very strange situation, I know... But this is all I was able to do.
Thanks

starbasessd, 2021-08-05:
You may have to install something like Ubuntu or other BaseOS, then install motionEye from one of the instruction sets here: https://github.com/ccrisan/motioneye/wiki/Installation
to get different results.

misanil, 2021-08-05:
Ok, I'll try this way.
In the meantime another issue comes out... 20190911 does not show images of one of the cameras.
I'm using the same settings as per previous version (this is the url used during configuration:    rtsp://1XX.YYY.WW.ZZZ/user=admin&password=password&channel=1&stream=0.sdp? ) and I'm sure it works since VLC still shows images with the same url
Funny enough...

starbasessd, 2021-08-05:
Try:
rtsp://1XX.YYY.WW.ZZZ/channel=1&stream=0.sdp?
Put username & password in the correct boxes, and select UDP for the bottom box when adding the camera.
Which camera is it?

misanil, 2021-08-05:
Is a cheap one... Escam QF910

misanil, 2021-08-05:
I'm using also other two cheap Escam cameras (QD300) and they still work with the same url

starbasessd, 2021-08-05:
iSpy database shows:
https://www.ispyconnect.com/camera/escam
rtsp://admin:admin@192.168.1.20/user=admin_password=tlJwpbo6_channel=1_stream=0.sdp
Which should translate to:
rtsp://ip_address/channel=1_stream=0.sdp with user & password in the correct boxes
and I would try both TCP (default) and UDP

misanil, 2021-08-05:
Whoooowwww!!!
With rtsp://1XX.YYY.WW.ZZZ/channel=1&stream=0.sdp? it works!!!
Thank you very much starbasessd!!!

misanil, 2021-08-05:
Very happy... After several days of my unuseful tests!

JamBalaya56562, 2026-11-10:
Closing as a duplicate of #1401: the grey picture with newer Chrome was fixed in motionEye 0.41 (motionEyeOS 20190911), and the camera that stopped after the upgrade works with the RTSP URL found above.

closed 2026-11-10 by JamBalaya56562