motioneye-project/motioneye #2155: h264_vaapi ffmpeg codec option unavailable

heavyinfo, 2021-07-09:
Greetings.

motionEye 0.42.1
Ubuntu 20.04
Intel Haswell CPU

I have recently switched the motioneye server setup from Raspberry Pi 3, ArchARM to x86_64 (Haswell) running Ubuntu 20.04 for performance reasons.

Even though my CPU supports intel vaapi and ffmpeg has it enabled the option was not available in the Movie Format although all other options for Hardware Acceleration such as OMX including the ones not applicable to my system such as NVENC, V4L2M2M are available.

H.264/OMX works fine, But I would like to use H.264/VAAPI instead for its better performance. I see that the option is excluded in the mediafiles.py code - https://github.com/ccrisan/motioneye/blob/795616d15caa24fd35f1d41b37668086e1ce4049/motioneye/mediafiles.py#L47-L60

I tried manually using the option in the config `movie_codec mp4:h264_vaapi` but doesn't seem to generate the video file.

Was VAAPI support removed for some reason I'm not aware of?

starbasessd, 2021-07-09:
Unknown, will leave here for a developer to answer.

zagrim, 2021-07-10:
I'd guess the question is more like has VAAPI support ever been there yet. But since I'm not too sure how all the things involved here work together in details, let's first check if it really even matters whether ME supports VAAPI or not:

You can run Motion by itself and see what happens. First shut down MotionEye (so that it doesn't interfere), then modify camera-X.conf to have `movie_codec mp4:h264_vaapi` and then run Motion manually with 
```
`which motion` -n -c /etc/motioneye/motion.conf -d 5
```
If that works and you get video files, then this is just a MotionEye issue (and perhaps rather easy to fix by just adding a line in that mapping you referred). However, I suspect that either Motion might be missing something, or even though you said VAAPI is enabled for you ffmpeg, in reality it doesn't work (well, that should also be possible to check by manually encoding some file on the command line, but I'm not that well familiar with `ffmpeg` options that I trust I'm able to provide detailed instructions for that right now).

starbasessd, 2021-07-10:
Thinking of your response, Esa, it could be his version of motion doesn't support it, but he doesn't mention which version is installed. He might want to upgrade motion to 4.3.2 just to be sure.

heavyinfo, 2021-07-10:
@starbasessd 
I apologize for not mentioning motion version it was 4.2.2, I upgraded it to 4.3.2-1 from motion releases.

@zagrim  You are right, Motion doesn't seem to be liking h264_vaapi as it failed to allocate codec

TL;dr
The correct option to use h264_vaapi for encoding is `movie_extpipe` and not movie_codec.

```
`which motion` -n -c /etc/motioneye/motion.conf -d 5
[0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
[0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
[0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-2.conf
[0:motion] [NTC] [ALL] motion_startup: Logging to syslog
[0:motion] [NTC] [ALL] motion_startup: Motion 4.3.2 Started
[0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[2:ml2:Front Camer] [ERR] [NET] netcam_rtsp_open_context: Normal resolution: Unable to find stream info: Success
[2:ml2:Front Camer] [ERR] [VID] vid_start: Netcam RTSP failed to open
[2:ml2:Front Camer] [WRN] [ALL] motion_init: Could not fetch initial image from camera 
[2:ml2:Front Camer] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
[2:ml2:Front Camer] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
[2:ml2:Front Camer] [ERR] [ENC] ffmpeg_set_codec: Could not open codec Invalid argument
[2:ml2:Front Camer] [ERR] [ENC] ffmpeg_open: Failed to allocate codec!
[2:ml2:Front Camer] [ERR] [EVT] event_ffmpeg_newfile: Error opening context for movie output.
curl: (7) Failed to connect to 127.0.0.1 port 8765: Connection refused
[4:nc4:Front Camer] [ERR] [NET] netcam_rtsp_handler_reconnect: Normal resolution: Reconnecting with camera...
```

But ffmpeg works fine for h264_vaapi, When tested with,

`ffmpeg -vaapi_device /dev/dri/renderD128 -i big_buck_bunny_720p_1mb.mp4 -vf 'format=nv12,hwupload' -c:v h264_vaapi output.mp4`

I noted that h264_omx is still blacklisted from the logs,

`[1:ml1:Back Camera] [WRN] [ENC] ffmpeg_set_codec_preferred: Preferred codec h264_omx has been blacklisted: Codec causes lock up on your FFMpeg version`

It seems that h264_omx automatically reverts to default - https://github.com/Motion-Project/motion/pull/1012

So even h264_omx is not being used in the media_codec.

So I checked ffmpeg.c of motion and didn't find h264_vaapi - https://github.com/Motion-Project/motion/blob/21e5bf90b5eeb6e0927fe9c9da13c872b77860da/src/ffmpeg.c

This lead me to search for other options and found users using movie_extpipe for HW accelerated encoding,

`movie_extpipe ffmpeg -y -f rawvideo -pix_fmt yuv420p -video_size %wx%h -framerate %fps -vaapi_device /dev/dri/renderD128 -i pipe:0 -vf 'format=nv12,hwupload' -c:v h264_vaapi %f.mp4`

I'm using that along with `movie_codec mp4`.

h264_vaapi can also be used with `netcam_decoder` if the camera supports it.

closed 2021-07-10 by heavyinfo

mspbonn, 2026-10-08:
`movie_extpipe ffmpeg -y -f rawvideo -pix_fmt yuv420p -video_size %wx%h -framerate %fps -vaapi_device /dev/dri/renderD128 -i pipe:0 -vf 'format=nv12,hwupload' -c:v h264_vaapi %f.mp4
`

fails in my case with

`event_extpipe_put: Fehler beim Schreiben in die Pipe, Statusfehler 1: Datenübergabe unterbrochen (broken pipe)`

My sources are the loopback devices `/dev/video0` and `/dev/video2`.

If I run the command

`ffmpeg -y -f rawvideo -pix_fmt yuv420p -video_size 2560x1440 -framerate 25 -vaapi_device /dev/dri/renderD128 -i /dev/video0 -vf 'format=nv12,hwupload' -c:v h264_vaapi -t 10 test.mp4
`

from a terminal everything is fine. What's going wrong ?