motioneye-project/motioneye #2156: Unable to change port from 8765 to 80

Shadal18, 2021-07-09:
I have Motioneye installed and running on Ubuntu 20.21. I have changed the config file in /etc/motioneye/motioneye.conf to reflect Port 80 and have rebooted the computer multiple times. The web interface still responds on port 8765 rather than port 80. What am I missing?

starbasessd, 2021-07-09:
I just did a: 
sudo nano /etc/motioneye/motioneye.conf
changed line:
port 8765 
to 
port 80
gave command:
sudo systemctl restart motioneye
and it immediately took effect
I then rebooted 
Worked after reboot.
What Ubuntu are you running? There is 18.04 LTS, 18.10, 19.04, 19.10, 20.04 LTS, 20.10, & 21.04
If not 18.04 LTS, 20.04 LTS or 21.04, I'd suggest upgrading...

Switched it all back reversing the process, also worked.

Shadal18, 2021-07-10:
That's what I've done. I'm running the latest version of Ubuntu, 21.
I have to assume it's because I restored from an MotioneyeOS backup that it's not obeying the correct .conf file, but I've grep'd the os and cannot find any other file containing 8765 related to Motioneye.

starbasessd, 2021-07-10:
MotionEyeOS uses port 80, so that shouldn't have been an issue.
motion/motionEye uses the /etc/motioneye/motioneye.conf and motion.conf files.
if you 
cat /etc/motioneye/motioneye.conf what does it show for port?
if you do:
sudo find -name motioneye.conf 
where does it find the filename?
I regularly do cross restores (backup from either motioneye or motioneyeOS) to the other, without any unexpected issues. 
What is in the file:
/etc/systemd/system/motioneye.service
mine shows:
-----
[Unit]
Description=motionEye Server

[Service]
ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
Restart=on-abort

[Install]
WantedBy=multi-user.target
-----
If you run sudo systemctl status motioneye
what does it report?

starbasessd, 2021-07-10:
Running as 'normal user'? Using HomeAssistant? Running anything els on port 80?
What does 
sudo netstat -apt 
show?

Shadal18, 2021-07-10:
I think I have other underlying issues possibly..

This was a clean install of Ubuntu 20.04, but now after having shut the system down for a while and booting it back up, I cannot even access motioneye via browser, even though Ubuntu isn't throwing any errors when starting the service..

I'm going to start with a fresh install and will get back to you shortly - thanks for your help!

starbasessd, 2021-07-10:
If you continue to have issues after a fresh install, please answer the questions from above.

Shadal18, 2021-07-10:
I must have something wonkey in my MotionEyeOS motioneye.conf file. I did a fresh install, restored my backup, copied the default .conf file back in and everything is working correctly now.

starbasessd, 2021-07-10:
Interesting. Glad to hear you're up and running. When you feel comfortable, remember to close the issue.

Shadal18, 2021-07-10:
Thanks again!

closed 2021-07-10 by Shadal18