motioneye-project/motioneye #2158: Buffering policy argument in open() function call (python3 branch)

takashikumagai, 2021-07-10:
(Commit on my forked repo: https://github.com/takashikumagai/motioneye/commit/75c68de80bad48c9831de9b3ed94b8d02baea88f)

To run motionEye as a non-root user with Python 3, I had to change

```python
        se = open('/dev/null', 'a+', 0)
```

to

```python
        se = open('/dev/null', 'a+')
```

in motioneye/server.py

Otherwise, the following error occurs and motionEye fails to start.

```
viridian@selkirk:~/src/motioneye-fork/motioneye (python3)$ python3 meyectl.py startserver -b -c /home/viridian/motioneye/motioneye.conf
viridian@selkirk:~/src/motioneye-fork/motioneye (python3)$ Traceback (most recent call last):
  File "meyectl.py", line 275, in <module>
    main()
  File "meyectl.py", line 255, in main
    server.main(arg_parser, sys.argv[2:], command[:-6])
  File "/home/viridian/src/motioneye-fork/motioneye/server.py", line 435, in main
    daemon.start()
  File "/home/viridian/src/motioneye-fork/motioneye/server.py", line 111, in start
    self.daemonize()
  File "/home/viridian/src/motioneye-fork/motioneye/server.py", line 74, in daemonize
    se = open('/dev/null', 'a+', 0)
ValueError: can't have unbuffered text I/O
```

So the question is, is there any reason to employ different buffering policies for stdin and stderr, respectively.
Here is the code in quesion in server.py:

```python
        # redirect standard file descriptors
        sys.stdout.flush()
        sys.stderr.flush()
        si = open('/dev/null', 'r')
        so = open('/dev/null', 'a+')
        se = open('/dev/null', 'a+', 0)
        os.dup2(si.fileno(), sys.stdin.fileno())
        os.dup2(so.fileno(), sys.stdout.fileno())
        os.dup2(se.fileno(), sys.stderr.fileno())
```

If the answer is no, I'll make a pull request against the python3 branch.

The error above occurs on 2 different systems shown below. So it's likely not an accident/something else in my environment.

---
Model: Raspberry Pi 3
OS: Raspberry Pi OS Lite (/etc/os-release is "Raspbian GNU/Linux 10 (buster)")
Linux kernel version: 5.10
Motion version 4.3.1+git20210125-34091a7
motionEye Version: 0.42 (python3 branch)
Python version: 3.7.3

---
Model: a mini pc with N3450 @ 1.10GHz (4 cores)
OS: Linux Mint 20 (based on Ubuntu 20.04)
Linux kernel version: 5.4.0
Motion version: 4.2.2
motionEye Version: 0.42 (python3 branch)
Python: 3.8.5

zagrim, 2021-07-10:
According to my experience it is rather common to have stderr unbuffered, I think that's some sort of a convention at least on *nix systems. The point, I assume, is that if the error stream was buffered, and the (application) process crashes, the error that occurred immediately before the crash might never end up being written in the log or console.

takashikumagai, 2021-07-10:
I see. Yes, that does make sense.

takashikumagai, 2021-07-11:
In Python 3, disabling buffering is only allowed in binary mode, so we have to come up with a workaround here.

For example, this seems to work too:

```python
        se = open('/dev/null', 'ab+', 0)
```

and **ab+** is for "opening a file for both appending and reading in binary mode."; I'm not sure if it's okay to change the code this way. Any suggestions on this?

zagrim, 2021-07-11:
Python documentation says only this about the difference between text and binary modes: `text mode [...] may convert '\n' characters to a platform-specific representation on writing and back on reading`. Since ME doesn't support other platforms than Linux, that shouldn't be an issue, so IMO it would be ok to use binary mode. But should it be used for all output, then? That's perhaps out of scope for this.

takashikumagai, 2021-07-14:
Sorry that my reply is rather late.
Do you mean that we should check other parts of the codebase as this redirection setting is applied to all the sys.stderr.write() calls? Any suggestions on what I can do to see if the change that I posted above is okay?

zagrim, 2021-07-15:
Yes, what I was referring to was other similar places in code but actually it might well be that there are none.

It would be better if someone with more practical experience with Python would take part in here. Perhaps @ccrisan could comment on this (and the related PR)?

takashikumagai, 2021-07-15:
Ah okay. I had done a quick searching myself yesterday on the master branch, and I found a few sys.stderr.write() calls, so I thought I should ask for advice. Is that what you meant?

As for the use of dup2, if I understand correctly, these are done to point both stdout and stderr to devnull to suppress all the output to the terminal? I searched the codebase for these too, and this is done when the motionEye is started in the daemon mode (-b). There is no other code in motionEye codebase that does this.

Also, after making a couple more changes (other than #2152 and this #2158), motionEye seems to be working with Python 3, but I've only used a minimal set of features (motion detection + notification). So let me know if there are features that are definitively popular and very frequently used.

MichaIng, 2022-03-12:
Solved in `dev` branch. Marking this as closed.

closed 2022-03-12 by MichaIng