motioneye-project/motioneye #2159: Motion says mp4:h264_nvenc is not supported. How does hardware acceleration work?

coen800, 2021-07-17:
I'm trying to use my GTX 1050TI to reduced the load on my cpu.

Ubuntu 18.04
Motioneye 0.42.1
Motion 4.0

First I installed the nvidia drivers and cuda SDK to compile ffmpeg with cuda/nvenc hardware accel. I did this by following [this build script](https://github.com/markus-perl/ffmpeg-build-script)
After confirming ffmpeg worked with hwaccel I installed motioneye according to the [install instructions for 18.04](https://github.com/ccrisan/motioneye/wiki/%28Install-On-Ubuntu-%2819.10-or-Earlier%29).
I added my camera and setup recording to disk with the h.264/NVENC movie format option. After a few seconds the camera previews stops and I find this in the motion logs.
```
[1:ml1:doorcam] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://x.x.x.x:554//h265Preview_01_main)
[1:nc1:doorcam] [ALR] [NET] netcam_handler_loop: Camera handler thread [3] started
[1:ml1:doorcam] [ERR] [ENC] get_oformat: ffmpeg_video_codec option value mp4:h264_nvenc is not supported
[1:ml1:doorcam] [ERR] [EVT] event_ffmpeg_newfile: ffopen_open error creating (new) file [/var/lib/motioneye/Camera1/2021-07-17/10-32-21]: Operation now in progress
[1:nc1:doorcam] [ALR] [NET] netcam_handler_loop: netcam camera handler: finish set, exiting
```
I also tried the HEVC/NVENC movie format but results in the same error `[1:ml1:doorcam] [ERR] [ENC] get_oformat: ffmpeg_video_codec option value mp4:hevc_nvenc is not supported`

If I understand correctly compiling ffmpeg with cuda/NVENC support is all that's needed to make it work according to this PR https://github.com/ccrisan/motioneye/pull/1708

starbasessd, 2021-07-17:
Upgrade to motion 4.3.2 (Bionic) found here:
https://github.com/Motion-Project/motion/releases/
Then test and report.
It may be that the version of motion you are trying to use has issues.
Also, are you running as root, or another user?

coen800, 2021-07-17:
Okay, will do.
Motioneye is running as root.

coen800, 2021-07-17:
Upgrading to motion 4.3.2 resolved the issue.

closed 2021-07-17 by coen800