motioneye-project/motioneye #2161: Can't have any SPACES in Text?

smyers75, 2021-07-21:
If I try to add any spaces to a Screen Text Overlay update, I get a pycurl error

Example

```
farhTemp=89.1   **#manually set for example**
humidity=40.2   **#manually set for example**

c = pycurl.Curl()
c.setopt(c.URL,'http://localhost:7999/1/config/set?text_left={}{}{}{}'.format(fahrTemp,"DEG_______",humidity,"RH"))
c.perform()
c.close()
```

The above code works, but if I change the underscores to spaces, as in **(fahrTemp,"DEG       ",humidity,"RH"))**
it errors out with:

Traceback (most recent call last):
  File "./readi2c.py", line 57, in <module>
    c.perform()
**pycurl.error: (52, 'Empty reply from server')**

Is this a Motioneye error, or something wrong with my pycurl?

starbasessd, 2021-07-21:
I don't know. I use the Text Overlay from the WebGUI as such:
%d %m %Y %H:%M
and it shows in the camera-1.conf file as:
text_right %d %m %Y %H:%M
and shows on the screen as:
21 07 2021 17:45
Have you tried to use escape characters "\ " or  "%20"?

smyers75, 2021-07-21:
CLOSED/Solved by changing the SPACES to %20 's as in
"DEG%20&20%20%20%20%20%20"

closed 2021-07-21 by smyers75