motioneye-project/motioneye #2162: Streaming resolution, no effect?

magnusso, 2021-07-24:
Should the streaming Image Resizing have an effect, on what specifically?

The quality of the video on MotioneEye user interface changes but there are no chanegs on the following:

Snapshot URL
Streaming URL
 Embed URL

starbasessd, 2021-07-24:
If you enable it, and change the slider to 100%, the outputs should be the same as the camera resolution set in Video Device. Set it to 50% and the outputs should be half of the resolution in Video Device, etc... The output URLs won't change.

magnusso, 2021-07-25:
@starbasessd, should the slider effect the size of the snapshot image URL, and the size of the video streamed on Streamed URL. The reason why I ask is that there is no effect. Any idea of what could be going wrong?

zagrim, 2021-07-25:
So when you mention URL, you don't mean the actual URL (the address), but what is received when streaming using that URL?

Anyway, it looks like that feature is only meant to affect the user interface. The "Streaming Image Resizing" switch has this help text (shown when hovering the question mark): "when this is enabled, the images are resized before they are sent to the browser (disable when running on a slow CPU)". The setting of the slider is only used when getting the "current" camera frame, and that doesn't seem to have any effect on the mentioned three URLs, only the UI. I guess the help text might be a bit more specific...

starbasessd, 2021-07-25:
In testing, you are correct, @zagrim as usual. I turned it on, and set it to 0% and it only affected the  WebGUI image, and I  slowly increased. It went from a complete blur, to slowly improving. I am not sure what the intent of the option was, but currently I don't think it's doing what was intended.

zagrim, 2021-07-25:
What would you expect the result to be when you downscale the image resolution to zero and then scale it back up to the original frame size? :laughing: 
I tested that too, and got a true flashback from the days of 4-bit palettes of the 80s. I don't really know why I got also this reduced colour palette, but the code calls Image.thumbnail from Pillow, and even though the Pillow documentation doesn't mention any restrictions or special cases for the size, it seems to do funny stuff when given a really low size. Maybe really small size is interpreted as a sort of an icon, and reducing the palette was seen as a useful addition there :shrug:  Anyway, I'd say that is a "feature" in PIL/Pillow, and ME just inherits that.

Anyway, looking at the help texts and the code, the streaming image resizing feature in ME seems to be meant to reduce the amount of data that is passed to the browser for each and every frame of the camera in the GUI. Setting it to something more sensible like 20% seems to do what it should. I don't think either that this is really useful feature as-is, but as I also just briefly tested running a browser session over ssh connection on a Raspberry Pi 3 and noticed a terrible latency due to bandwidth and CPU limitations, I'd assume that this sort of setup might benefit from downscaling the stream frames. Not very common setup though.

starbasessd, 2021-07-25:
In my little pea-brain, I would think Setting resolution to 50% would reduce the size of the image to 1/2 of the primary. IOW if the primary stream is 800x600, it would reduce it to  either 600x400 (1/2 the area) or 400x300 (1/2 the dimensions) not keep the same image size but increase the pixelisation (reduce the bandwidth information). But maybe that's me. Otherwise, it seems to me, you are just effectively reducing the jpeg quality function...
BTW, thanks for responding, you always bring something good to think about...

starbasessd, 2021-07-25:
Another BTW. if you have various cameras, with image sizes from, say 640x480 to 1280x720, and you want to try to equalize the image sizes, the Streaming Image Resize could be a neat feature, if it was like I described above. I have that issue, and have to place my 1280 cameras at the end so there isn't a lot of dead space in the middle of the all camera view.

zagrim, 2021-07-26:
It will reduce the dimensions according to the setting, so in your example the resulting resolution would be 400x300. However, as the GUI shows the camera frames with equal widths, the size of the camera frame in the GUI doesn't change, only the picture quality degrades with pixels getting visible. Basically the browser is stretching or squeezing the image as necessary so that it fits to the block of screen estate that is reserved for it.
I just tested with two streams from same camera with resolutions 1280x960 and 704x576, they are shown with equals widths and only the height varies with the aspect ratio (4:3 and 11:9 (don't know why it has such a weird resolution)). I believe also you @starbasessd have those 640 pixels wide camera images show with equal width as those that are 1280 pixels wide?

zagrim, 2021-07-26:
As for the latter comment @starbasessd , if you have both 16:9 and 4:3 aspect ratios and would like to equalise them so that also their displayed heights would be the same (possibly squeezing or stretching vertically?), that would require the GUI layout (HTML/CSS) be done somewhat differently, I think... I mean the fitting of the image to the space reserved for it is done solely by the browser based on the CSS styles, and even though one would think that that sort of thing would be easy, but unfortunately it's usually not :disappointed: But then again, we've never been friends, me and CSS...

starbasessd, 2021-07-26:
Ahhh. CSS. The bane of users everywhere. That does explain my issues. That's why I'm not a Web Dev, and have a few on my 'hit' list... Thanks again for your input...

magnusso, 2021-08-29:
Thanks for the insights! Perhaps a silly question, but why is not the image produced by the snapahot URL affected, i.e. the first link. The reason why I ask is that I'm tryning to thet a thubmnail, say 300x300, size image to Node Red. Yes, I can resize in Node Red, but with a cameras between 8MP-12MP resizing a single image in Node Red takes between 30-60 seconds.

zagrim, 2021-08-29:
@magnusso the short answer is that it's not designed to work so :smiley: I mean, it's designed to affect only the frames in the camera video stream MotionEye server sends to the user interface (it might not be obvious, but ME server is not actually serving a video stream to the UI, but a stream of frames - not that it really makes any difference regarding what you asked).

I've not quite figured out how the request handling works inside of ME, but it seems to me that if it just was possible to alter the snapshot URL without causing "unauthorized" error, one might even be able to pass a "width" and "height" parameters with it (but I might be mistaken and the request handler method gets its parameters not directly from the URL but from some other method somewhere in the code). I seem to vaguely remember someone here once explaining how to produce a valid "_signature" parameter for a given URL, but I'm not at all sure that really happened. Anyway, what you are asking might not be too hard to implement given there was someone with the skills, time, energy and motivation to do it (myself I'm lacking mostly time and energy to figure out those things I don't yet know about how ME works internally).

JamBalaya56562, 2026-11-10:
Closing, as zagrim answered this: Streaming Image Resizing only shrinks the frames motionEye sends to its own UI (`picture/<id>/current/`); Motion's stream on the streaming port is not resized by it.