motioneye-project/motioneye #2166: on/off 'Motion Notification' via shell command

Pedder007, 2021-07-28:
Hello everybody,

I already searched around on the opportunity, to switch on and off the Motion Notification function in motioneye via shell command, like one can also start and stop Motion Detection by sending e. g. http://192.xxx.xxx.xxx:7999/1/detection/[start/stop].
As I just verified, that if one switches the notification off manually, motioneye forgets the respective config parameters, for sure then they would have to be stored somewhere temporary.

Is there an opportunity to have such in motioneye?

Kind regards
Peter

starbasessd, 2021-07-28:
Flagged for 'feature'/'enhancement'.

starbasessd, 2021-07-28:
One option is to set up 2 copies of the camera-x.conf file and follow the instructions by calling the files camera-1.conf.email and camera-1.conf.no-email...
https://github.com/ccrisan/motioneyeos/issues/2779#issuecomment-882773284

Pedder007, 2021-07-29:
Thanks a lot @starbasessd, I will test your idea!
Background is to switch on and off the notification automatically, indepently from the motion detection, when leaving the house. If beeing at home, for sure no mail notification is necessary ;-)

Pedder007, 2021-07-29:
ok, in general works, but have to tackle that I can run the script also from another pi
will check this the next days
thanks again!

starbasessd, 2021-07-29:
You can use SSH to run a command on a remote Pi.
https://stackoverflow.com/questions/53011738/execute-on-raspberry-pi-using-ssh-a-command-that-let-run-a-py-script-on-another

Pedder007, 2021-07-30:
well, in the end I failed 
I followed your link but also others like this (https://www.raspberrypi.org/documentation/remote-access/ssh/passwordless.md), a bit more detailled, but in the end wasn't able to connect without typing in a password each time.
I tested and tested and spent hours of research, but wasn't able to change this. I have no clue why this stays but also found others struggeling with this but no working solution ...

Thus hoping now for a direct shell command feature here ...

starbasessd, 2021-07-30:
Possible with motionEyeOS, not so much on another OS. Let me get you some instructions.

Pedder007, 2021-07-30:
I'm running motioneye on raspbian buster on a PI4B. Reinstalling it completely would take me some more time....
If you have any ideas on the ssh login, please share.
I just in addition checked/changed the sshd_config but that also didn't change the behaviour...

RSAAuthentication yes
PubkeyAuthentication yes
AuthorizedKeysFile      %h/.ssh/authorized_keys

It's late here now, I will close for today ...

starbasessd, 2021-07-30:
On the pi you are going to run the scripts from:
ssh-keygen
save the file to /home/pi/.ssh/id_rsa
ssh-copy-id -i ~/.ssh/id_rsa [ip_address_of_Pi_with _the_scripts]
now if you ssh to the Pi with the scripts, you shouldn't be prompted for password
If that is true, type exit on the screen, should bring you back.
Now try your scripts (full path needed)
ssh [ip_of_script_pi] sudo /path/to/script
I hope you backed up the sshd_config file, no need to make changes.
If you need/want other options, let me know...

Pedder007, 2021-07-31:
Hi again, thanks for your response.and guidance.
I yesterday did exactly what you wrote and when I now try to 'ssh-cops-id' again, I get the answer that the key is already available on the remote Raspi:
````
/usr/bin/ssh-copy-id: INFO: Source of key(s) to be installed: "/home/pi/.ssh/motionmail.pub"
/usr/bin/ssh-copy-id: INFO: attempting to log in with the new key(s), to filter out any that are already installed

/usr/bin/ssh-copy-id: WARNING: All keys were skipped because they already exist on the remote system.
		(if you think this is a mistake, you may want to use -f option)
````

I additionally took back my changes in the sshd_config and also restarted ssh service, but the password is requested further on
The script itself on the remot client works, I already tested that. My only problem is the ssh login without a password.
And I definetly created the file (three times) without a password ;-)

starbasessd, 2021-07-31:
Ah! can't go and change the file names. Needs to stay the same. please try again, but use the commands as written. When it creates the 'authorized_keys' file, it tags each key in there with the user and computer it works for at the end of each key...
cat ~/.ssh/authorized_keys
ssh-rsa \<LONG string of seemingly random characters\> pi@your_pi_name
Let me know...

starbasessd, 2021-07-31:
BTW, if you remove the tag at the end, it makes it available for any machine to use that key. \(handy for using several Win10 PCs with PuTTY to connect...\)

Pedder007, 2021-07-31:
hmhm, I bit confused now.
the file 'authorized_keys' I only do have in the .shh folder on the target pi
thus I assume that I have to execute the commands from there, right?
and the <LONG string....> is what I see in that file when I open it with an editor, right?
I already struggled yesterday with the question, from which side (master or target) I have to execute the related commands ...

Pedder007, 2021-07-31:
ah sorry, git you wrongly.
For sure 'cat' shows the content and the result is what you described above with 'ssh-rsa LOG....'
Now I realized what you meant by taking the same file sames. I thought that it is regardless how the file is called, as along as I use the same when creating and then for the 'ssh-copy-id' command ....

starbasessd, 2021-07-31:
No, as to having to run from the .ssh folder, ssh username@ipaddress can be run anywhere from a terminal. Ditto for ssh username@ipaddress command_to_run
To create the keys, you do so on the Pi you want to connect from, where the scripts will be run from, and the ssh_copy_id -i ~/.ssh/id_rsa [ip_address_of_Pi_with _the_scripts] copies the needed stuff to the right place.

Pedder007, 2021-07-31:
OMG - it works!!! 😄 
I simpley chosed now 'id_rsa' as filename and that's it! never thought that taking an other name would have such an impact.
Thanks a lot again, I will now test to start the ssh command out of iobroker, to motioneye

starbasessd, 2021-07-31:
If iobroker runs as Pi it will work. If iobroker runs as root or another user, then you probably will still have issues. Let me know.

Pedder007, 2021-07-31:
ok, your question seems to be meaningful
from the console of the iobroker pi I can run:
ssh '192.168.xxx.xxx' /etc/motioneye/motionmailoff.sh
and also
ssh 'pi@192.168.xxx.xxx' /etc/motioneye/motionmailoff.sh
But both doesn't work when calling it from iobroker/blockly:
[image]

starbasessd, 2021-07-31:
Try:
sudo -u pi ssh pi@192.168.xxx.yyy /etc/motioneye/motionmailoff.sh
make sure the iobroker user is in group sudo with the command:
getent group sudo
should list at least pi

Pedder007, 2021-07-31:
no sorry, does not work

'getent group sudo' does prompt pi

starbasessd, 2021-07-31:
List the users:
sudo cat /etc/shadow
do you see an iobroker user? (they are not in alpha order)

Pedder007, 2021-07-31:
yes iobroker is listet there

in parallel I spent some research in the iobroker forum and found similiar problems from other users there
the solution seems to be, that one has to creat the respective key (ssh-keygen, .. etc) in the context of the iobroker user

starbasessd, 2021-07-31:
Or maybe run:
ps aux | grep iobroker
and report what's in the first column.

Pedder007, 2021-07-31:
shows a long list which adapters iobroker seems running
the first column always shows 'iobroker'

starbasessd, 2021-07-31:
Try doing what your research showed to get a ssh key for iobroker. Probably won't work because iobroker is a system account. 
ssh-keygen
save the file to /home/iobroker/.ssh/id_rsa
ssh-copy-id -i ~/.ssh/id_rsa [ip_address_of_Pi_with _the_scripts]
If you can, then change your command to:
ssh pi@192.168.xxx.yyy /etc/motioneye/motionmailoff.sh
Might have to step back and re-think...

Pedder007, 2021-07-31:
it seems to become an endless story ... ;-)
````
pi@raspberrypi:~ $ sudo su -P iobroker -s /bin/bash
iobroker@raspberrypi:/home/pi $ ssh-keygen
Generating public/private rsa key pair.
Enter file in which to save the key (/home/iobroker/.ssh/id_rsa): id_rsa
Enter passphrase (empty for no passphrase): 

Enter same passphrase again: 

Saving key "id_rsa" failed: Permission denied
````

starbasessd, 2021-07-31:
Does the path exist?
sudo su -i
cd /home/iobroker
and
cd /home/iobroker/.ssh
???
if not, then make them:
mkdir -p /home/iobroker/.ssh
chown -r iobroker:iobroker /home/iobroker
exit
then try your keygen commands again.

Pedder007, 2021-07-31:
yep, is there

starbasessd, 2021-07-31:
If you do ls -la /home/iobroker what are the owner and group?

Pedder007, 2021-07-31:
drwxr-xr-x 7 iobroker iobroker 4096 Jul 31 15:23 .
drwxr-xr-x 4 root     root     4096 Mai 11  2020 ..
-rw------- 1 iobroker iobroker   27 Jul 31 15:34 .bash_history
-rw-r--r-- 1 iobroker iobroker  220 Apr 18  2019 .bash_logout
-rw-r--r-- 1 iobroker iobroker 3523 Feb 13  2020 .bashrc
drwxr-xr-x 3 iobroker iobroker 4096 Mai 11  2020 .cache
drwx------ 3 iobroker iobroker 4096 Mai 11  2020 .config
-rw-r--r-- 1 root     root       18 Jun 15 21:00 iobrok_data.db
-rw-r--r-- 1 root     root       17 Jun 15 21:03 iobrok_data.sqlite
drwxr-xr-x 2 iobroker iobroker 4096 Nov  1  2020 .lgtv2
drwxr-xr-x 6 iobroker iobroker 4096 Jul 26 22:56 .npm
-rw-r--r-- 1 iobroker iobroker  807 Apr 18  2019 .profile
drwx------ 2 iobroker iobroker 4096 Jul 30 22:05 .ssh

Pedder007, 2021-07-31:
ok, I'm called here for some other stuff (family ;-) ) 
it might be the case that I'm able to continue here at least tomorrow evening
thanks for now and 'see' you :-)

starbasessd, 2021-07-31:
I will cogitate a while. I understand about family & life. Enjoy.

Pedder007, 2021-07-31:
Hi, I found another hour in the evening and followed up with the respective chat in the iobroker forum.
At least it went as following:
1. iobroker (or other user) entry in passwd has to be changed from:
    - iobroker:....../home/iobroker:/usr/sbin/nologin  to
    - iobroker:....../home/iobroker:/bin/bash
2. one has to switch to user iobroker to set up the key under his account
3. then ssh-keygen (as user iobroker)
4. the key then has to be copied to the remote-client by:
     - cat ~/.ssh/id_rsa.pub | ssh pi@192.168.xxx.xxx 'cat >> .ssh/authorized_keys'
5. the shell command in blockly then has to look like this
     - ssh pi@192.168.xxx.x /etc/motioneye/motionmailon.sh

PS.
That works propperly, but having a direct shell command via port 7999 in the end would be much more convenient.
When e. g. setting up one of the systems newly, I aks myself if I would remember all that. Surely I made my notes in my personal documentation ...
Thanks again for your support 👍

starbasessd, 2021-07-31:
No problem.
As a security issue, should you change it back to nologin?
As to the port 7999, that is motion's port, notification is a motionEye script...

Pedder007, 2021-07-31:
you are right, I will switch iobroker's rights back 👍 
regarding the port: aha ok, I thought it can be used for both ...

starbasessd, 2021-07-31:
No problem. The WebGUI is on port 80 or 8765 (motionEyeOS or motionEye). There isn't a command port for it. 7999 is for controlling motion. try it with http://ipaddress:7999, too.

Pedder007, 2021-07-31:
I konw, I'm already using this way for switching motion detection in general on and off.
That was why I was asking in this direct...

starbasessd, 2021-07-31:
Motion DETECTION can be controlled with 7999, it's a motion function. I think that is what you are saying that you understand.

Pedder007, 2021-07-31:
yep 😄, sorry I'm not that deep in and thus maybe do not differentiate propperly.

I stepped in that whole field of open source home automation and video surveillance now only two years ago and mostly only have the evenings to bring it forward. The entire field is very cool and flexible, but not that easy to tackle for one who has a solide technical and IT understanding, but is e. g. not a developer or a linux expert ... ;-)

It's getting late again here, take care and bye for now 😄