motioneye-project/motioneye #2167: Should I upgrade to Motion 4.3.1?

rocket59, 2021-08-03:
Currently on motioneye server 0.42.1. 

Was wondering if there was a benefit to also updating Motion from the default 4.2.2 to the latest stable release?

starbasessd, 2021-08-03:
4.3.2 is stable for me.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

rocket59, 2021-08-03:
Thanks - is there any major improvement over 4.2.2?

starbasessd, 2021-08-03:
The changelog is here.
https://github.com/Motion-Project/motion/releases/tag/release-4.3.2
4.3.2 has been out since October 2020. They've not been keeping up with their dev schedule, so unsure if it's just very stable, or no updates past 4.3.2 needed.

closed 2021-08-03 by starbasessd