motioneye-project/motioneye #2168: No camera, no ssh

AeroChaker, 2021-08-03:
Hello,
I installed motioneye on a raspberry pi 3 to watch my cats and went on vacation, worked great for several days.
But one day, I've got no camera, I looked all over the internet for a solution but no luck.
I tried to connect through SSH, but I forgot to fix the port redirection.
I think my system partition was flooded with temp files or something like that.
I still have access to motioneye, but I can't find the real syntax to reboot from it.
I'm sorry for my bad English and bad explanation. I'm not a tech expert.
I just need to reboot it with "run a command" if possible.
Thanks in advance and sorry again.

starbasessd, 2021-08-03:
MotionEye or motionEyeOS?
If motionEyeOS, the reboot is in Settings, Expert Settings.
If you are running motionEye on another OS you would SSH to the base OS and run reboot there. If you try "Run A Command" it's in Settings, Motion Notifications,
You'd have issues if you are running motionEye as a non-root user, but I'd try:
sudo reboot
and the next time it detects motion it would trigger. But if the camera doesn't appear that's probably not going to work.
but if the system crashed due to disk drive full, it probably won't reboot correctly. It wouldn't be tmp files. It might be log files, but I doubt it. It's probably video or pic files if you didn't save them to a SMB NAS/share. 
I believe your current only option is to physically get to where you can SSH in or use the console  and reboot.

starbasessd, 2021-08-03:
@ccrisan Feature Enhancement: Enable "enable_reboot true" by default in /etc/motioneye/motioneye.conf by default, which also enables "Shutdown" and "Reboot" buttons in Settings, General. The Comments lines only mentions wifi settings and time zone, which is only effective in motionEyeOS.

AeroChaker, 2021-08-03:
Thank you for the fast answer.
I'm running motioneye not motioneye os on dietpi.
With smb storage, I think its the log files as you said.
I think the only solution is the SSH, but I'm over 2000km away from home.
I tried several commands in the "file storage > run a command " section, but won't work.

starbasessd, 2021-08-03:
Try it under Motion Notifications, Run A Command, but if the camera is missing, that's a problem. Also if you are using File Storage, Upload to a SMB share, then you are creating the movies and pics in 2 locations, locally and on the share. I can walk you through a better way when you get control back.

AeroChaker, 2021-08-03:
When I edit something :
[image]
Dietpi storage, SD card
[image]
Default storage, SD card 64gb
[image]
SMB storage 
[image]

As you see, in the default storage, it looks too small. 
The problem percist in all 3 storages.

starbasessd, 2021-08-03:
That's because the / tree is full, I'd bet.
For grins and giggles, if you can afford to lose your current pics and vids, you could set 'preserve' (under movies and still images) to 1 day. Come back in 25 hours, and see if you can access your camera. Of course if you don't have that option...
You are between a rock and a hard place...

AeroChaker, 2021-08-03:
That's what I thought too, I tried to delete some files but I lost access to them too when I deleted and reinstalled the camera.
Thanks again for your time, and I hope to regain SSH access.

AeroChaker, 2021-08-08:
Update: I managed to send someone to my house to shutdown the electricity and turn it back on.
I was hoping that the log files will be deleted, but no luck, I lost even control of the settings now, unlimited loading.
I'll keep trying to gain SSH Access.

starbasessd, 2021-08-08:
Definitely points to Disk Full, whether logs, pics, or something else.

AeroChaker, 2021-08-16:
Finally, I gained access to SSH, I'm doing a log clean up.
I'm using Dietpi.

starbasessd, 2021-08-16:
look for the .jpgs, too.

AeroChaker, 2021-08-16:
I didn't found any media. I rebooted, now the raspberry don't connect. I give up, bad luck.
Thank you again for the help.

starbasessd, 2021-08-16:
If you run into further issues, please go ahead and re-open or start a new issue.

closed 2021-08-16 by starbasessd