motioneye-project/motioneye #2169: [Feature request] Option to play sound in Web GUI

HDefectus, 2021-08-06:
Hi,

I'm using motion eye on a RPI ZeroW and have a laptop in my workshop to monitor if any customers come in the door. 
Would love if there was an option for the Web GUI (on the client) to play a chime when motion is detected.

MagicPlants, 2021-08-15:
That's what "Run a command" is for. Write a quick "aplay" based script. Bob's your uncle. Any further trouble just say the word.

starbasessd, 2021-08-15:
He was requesting the sound play IN the webGUI, not through an external speaker. We had that discussion in another forum, and I sent him over here to request the feature. Your solution only plays it locally to the motioneye doing the detecting, he wanted it to make noise on any computer connected to the WebGUI.