motioneye-project/motioneye #2170: Failure network Local  not found (under docker installation)

caslor, 2021-08-07:
Hi
i used a fresh image of Armbian Buster to burn and run it to my OrangePi zero  single board
with :` uname -a `
i get this :  `Linux orangepizero 5.10.43-sunxi #21.05.6 SMP Mon Jun 21 15:07:45 UTC 2021 armv7l GNU/Linux`

i installed :  docker engine , portainer,  homeassistant    
all working fine 

i used this command to install motioneye  : `run --name="motioneye" --network="local" --device=/dev/video0 -p 5000:8765 -p 465:465 -p 587:587 -p 8081:8081 --hostname="motioneye" --restart="always" --detach=true ccrisan/motioneye:master-armhf`

it installed the container but  under portainer  show it as " created"   and when i select it and press start  i get the message : **Failure network Local  not found**

any help?   (i am very new to docker thing just following some online instructions )

caslor, 2021-08-07:
ok i found the solution...  make it work

so i had to remove some commands from  the previous command line and changed the  `--network="local"`  to  `--network="bridge"`

working command line : `docker run --name="motioneye" --network="bridge" -p 5000:8765 -p 465:465 -p 587:587 -p 8081:8081 --hostname="motioneye" --restart="always" --detach=true ccrisan/motioneye:master-armhf`

starbasessd, 2021-08-07:
You can remove -p 465:465 -p 587:587 also, those are only needed if you were to be expecting INBOUND 'secure email' to the docker container, and you can't. Outbound is not affected.
There are known issues with HomeAssistant and Portainer wrappers, also. The recommended starting command is found here:
https://github.com/ccrisan/motioneye/wiki/Install-In-Docker
docker run --name="motioneye" \\
    -p 8765:8765 \\
    --hostname="motioneye" \\
    -v /etc/localtime:/etc/localtime:ro \\
    -v /etc/motioneye:/etc/motioneye \\
    -v /var/lib/motioneye:/var/lib/motioneye \\
    --restart="always" \\
    --detach=true \\
    ccrisan/motioneye:master-armhf
You can change the external port as needed  as you did.

caslor, 2021-08-07:
thanks !!

i will follow your suggestions

closed 2021-08-07 by starbasessd