motioneye-project/motioneye #2171: Generate movies/pictures thumbnail in the background / regularly

2Belette, 2021-08-09:
Hi,

When I have a long list of events in the same day it takes long to create the thumbnails and when I open the list I have to wait ~ 3sec for each thumbnail to be created for the first time (if I come back it is very quick).

Would it be a way to create them in the background let's say every hour or so ?

Thanks

starbasessd, 2021-08-09:
You don't state which version of motion, motioneye, or your base OS (or if using docker, HA, or other container). The thumbnails for the movies are created with the movies, not on first access. If you have a large number, it may take your browser some time to cache them. I run 6 cameras in production, and all folders create 2 files: (time.mp4 and time.mp4.thumb) when the movie is created.

2Belette, 2021-08-09:
Sorry :
motionEye Version 	0.42.1
Motion Version 	4.3.2
OS Version 	Parabola

The wait time is only for the first time I am asking for the daily list, after everything is loaded then it is fast so my question..I am surprised that the thumbnail is created with the video, I would expect it to be quicker to show

starbasessd, 2021-08-09:
How many files are in the folder?
Which browser are you using?
If you clear the cache in the browser, does it take a long time to display, again?
Do you have another PC or MAC and does it happen there?

warlon, 2021-09-10:
I also have this issue. I have a raspberry taking pictures from an IP camera every 10 seconds and it generates thousands of pictures every day. Opening the picture browser takes a very long time. Not sure what it is doing at that point, but my guess is generating thumbnails.

starbasessd, 2021-09-10:
It doesn't generate thumbnails on the server for jpgs. The images are re-sized on the client browser side.  Thumbnails are only generated on the server side for the movie files.

warlon, 2021-09-10:
Ok, it must be something else then. The part that takes long is when I first click the "open pictures browser" icon and the spinning thing appears. I have pictures stored for 7 days and folder for each day contains over 8000 pictures.

starbasessd, 2021-09-10:
Yep, and it goes through all 56000 pictures every time. It would take a while to get every image and shrink it down to a thumbnail.

warlon, 2021-09-10:
In the next step it shows me what subfolders I have and how many pictures is in each. I have one folder for each day. I don't think it would need to go through them all at that point.

starbasessd, 2021-09-10:
Shouldn't, but does. Delete or move every day but the current to another folder tree and see how much the speed improves.

warlon, 2021-09-10:
Sure it will be faster, but that's not really an option since I need this for surveillance. Guess I need to find some other solution for browsing the images.

starbasessd, 2021-09-10:
Easy enough to write a script to move them daily to something like /media/surveillance, or turn on SMB or FTP Share and view with  Networking File Manager or similar from another machine. I use the former method to move them to a NAS and then I delete them from there when needed.

warlon, 2021-09-10:
I need to view the files over the internet so that limits my options a bit. Also they have to be visible from a browser to less tech savvy end users. I was thinking about looking into some web gallery apps. I bet there are already many solutions to something like this.

2Belette, 2021-09-10:
@starbasessd sorry I just realized I miss your message and responding to your question from Aug 9th, my bad. Reading @warlon I am on the exact same situation recording one picture every 10sec to produce some timelapse video and keeping 1week so having thousands and thousands of pics per day. Opening the pictures folder from Motioneyes GUI is very very slow, what I don't get is why it taking so long as well for video where I use to have let's say less than 10 movies per day and the spinning is only the first time I access the folder after that it is almost instantly. I still think that the thumbnails for video is done from the first time I try to see the list and then it is cached and not when the movie is created..

starbasessd, 2021-09-10:
@2Belette Test it:
Wait for some videos to be recorded.
DO NOT view them through the WebGUI
SSH or open a console into the PC or Pi running motionEye
Go to the folder where you are saving the movie & jpg files
You will find movie file names, and matching files with the same name + .thumb
You will not find matching jpg + .thumb files
motionEye only generates movie thumbnails when it generates the movies.
It generates all other mini images on the fly on the client web browser. I do not 
write HTML or CSS code, but you may be able to go through the .py or .pyc code to 
see how it's done in motionEye, and maybe suggest better ways.

starbasessd, 2021-09-10:
BTW, the standard location for the files in motionEye is in /var/lib/motioneye/CameraX (X=camera number)

closed 2024-07-25 by 2Belette