motioneye-project/motioneye #2172: Retain function for motion detection

pacoso81, 2021-08-10:
Hi.
i'm using MotionEye as addon in HA.
i'm controlling motion detection from HA by rest_command try webcontrol.
My problem is that every time that MotionEye restarts, motion detection is going Active even if it was on Pause before restart.
Can you add retain function please.

starbasessd, 2021-08-10:
I believe the request is for 'retain last state upon restart'.
If the 'pause' command came through HA motionEye wrapper talking directly to motion, (port 7999) I don't believe it will be possible, that would be a HA/motionEye wrapper issue. If it is another function directly in HA, you might need to talk to them (HA Community). You could also ask the motion-project developers to consider such an option, too.

pacoso81, 2021-08-11:
Yes, you are right about the request.
Yes, the command is coming through port 7999, but i can't understand why it's not possible.
You already have the setting in the MotionEye already set, just need to retain it.
If i do it from the HA it will be very complex and time consuming, and i'm not very good in YAML.

starbasessd, 2021-08-11:
Because sending a command to port 7999 talks directly to motion, rather than using motionEye. It is NOT an option of motionEye. 
MotionEye  is port 8765, the UI wrapper that gives you the WebGUI. If you disabled motion detection there it would be in option Settings, Motion Detection \[On\] or \[Off\] . There is no indication of \[Paused\]. MotionEye doesn't know or sense that you paused detection in motion directly.